embassy-rs/embassy #3260: examples/rp fails to build; missing in ci-nightly script.

bsodmike, 2024-08-16:
Just tried compiling from within the embassy repo.  All defaults picked up as expected on nightly `rustc 1.82.0-nightly (2c93fabd9 2024-08-15)`

```rust
   Compiling aead v0.5.2
error: cannot find macro `impl_decrypt_in_place` in this scope
   --> /Users/mike/.cargo/registry/src/index.crates.io-6f17d22bba15001f/aead-0.5.2/src/lib.rs:310:9
    |
253 | macro_rules! ximpl_decrypt_in_place {
    | ----------------------------------- similarly named macro `ximpl_decrypt_in_place` defined here
...
310 |         impl_decrypt_in_place!(self, nonce, associated_data, buffer)
    |         ^^^^^^^^^^^^^^^^^^^^^ help: a macro with a similar name exists: `ximpl_decrypt_in_place`
```

This stems from https://crates.io/crates/reqwless and it's dependency on https://crates.io/crates/aead

## ci-nightly.sh

https://github.com/embassy-rs/embassy/blob/main/ci-nightly.sh#L29 is not running the `examples/rp` for builds.

Dirbaio, 2024-08-16:
I can't reproduce this. Try `cargo update` or removing Cargo.lock. Either way it's probably a bug in the `aead` crate, nothing we can fix here.

bsodmike, 2024-08-16:
What version has your lock for `aead` got?  I'll try again in the morning and update.

bsodmike, 2024-08-17:
> I can't reproduce this. Try `cargo update` or removing Cargo.lock. Either way it's probably a bug in the `aead` crate, nothing we can fix here.

FYI it's due to `reqwless` -> `embedded_tls` -> `aead`.  I can get the basic blinky to compile by removing this dependency, but of course, web based examples need it.

I've removed the lock as well BTW.

closed 2024-08-17 by bsodmike