embassy-rs/embassy #3265: get_level followed by wait_for_falling_edge returns high

overflowz, 2024-08-17:
Hi, I got this use-case which I'm trying to do, but I got this weird thing. I'll provide minimal poc here:

```rs
#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    let p = embassy_stm32::init(Default::default());
    info!("Hello World!");

    let d0 = Input::new(p.PC10, Pull::Up);
    let mut d0 = ExtiInput::new(d0, p.EXTI10);

    loop {
        let _ = d0.wait_for_falling_edge().await;
        info!("level: {}", d0.get_level()); // always returns High
    }
}
```

My expectations would be that whenever I get the falling edge event triggered, `get_level` should return the same value, but it does not. Is this expected behavior and I'm missing something?

The exact same thing worked fine on STM32G071RB, while it does not work on STM32L452RE.

Thank you.

Dirbaio, 2024-08-17:
If the low pulse is very short, by the time you do `get_level()` the pin could've come back to high.

Try building with --release and setting the RCC config so that the chip goes as fast as possible. This'll reduce the chances of this happening, but it's still not guaranteed if the pulse is reaaally short.

overflowz, 2024-08-17:
That could be it, the pulse comes as burst (lows and highs) and probably it does not catch it that fast.

Thank you for providing the direction (I'm still new), I'll play around and report back if that is the case.

overflowz, 2024-08-17:
Closing this issue. I'm not even sure how to increase the performance (I already borked one nucleo board), but from what I checked, it's exactly what you mentioned.

Thank you so much!

closed 2024-08-17 by overflowz

overflowz, 2024-08-18:
If anyone needs, it was indeed a clock issue. I boosted clock to 80MHz and it worked flawlessly (by default it was running on 4MHz):

```rs
#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    let mut cfg = Config::default();

    cfg.rcc.hsi = true;
    cfg.rcc.mux = ClockSrc::PLL1_R;

    cfg.rcc.pll = Some(Pll {
        source: PllSource::HSI,
        prediv: PllPreDiv::DIV1,
        mul: PllMul::MUL10,
        divp: Some(PllPDiv::DIV7),
        divq: Some(PllQDiv::DIV2),
        divr: Some(PllRDiv::DIV2),
    });

    let p = embassy_stm32::init(Default::default());
    info!("Hello World!");

    let d0 = Input::new(p.PC10, Pull::Up);
    let mut d0 = ExtiInput::new(d0, p.EXTI10);

    loop {
        let _ = d0.wait_for_falling_edge().await;
        info!("level: {}", d0.get_level()); // returns Low
    }
}
```