embassy-rs/embassy #3278: BufferedUart + half duplex

BKSalman, 2024-08-23:
Hello,

I wanted to have uart with `embedded_io_async::Write` & `embedded_io_async::Read` implemented, but also be half duplex

so I'm wondering if that is possible, and just hasn't been implemented yet, or not

thank you for the great project :)

badrbouslikhin, 2024-08-27:
Hi @BKSalman,
It's already possible. Take a look at [UartRx in embassy_stm32::usart - Rust](https://docs.embassy.dev/embassy-stm32/git/stm32f446re/usart/struct.UartRx.html#method.into_ring_buffered).

Here's a snippet of how to achieve this:
```rust
    const DMA_BUF_SIZE: usize = 256;

    let mut usart = Uart::new_half_duplex(
        p.UART4,
        p.PA0,
        Irqs,
        p.DMA1_CH0,
        p.DMA1_CH1,
        config,
        HalfDuplexConfig::PushPull,
    )
    .unwrap();

    let (mut tx, rx) = usart.split();
    static mut DMA_BUF: [u8; DMA_BUF_SIZE] = [0; DMA_BUF_SIZE];
    let dma_buf = unsafe { DMA_BUF.as_mut() };
    let rx = rx.into_ring_buffered(dma_buf);
```
In this case, `tx` implements `embedded_io_async::Write` and `rx` implements `embedded_io_async::Read`.

BKSalman, 2024-08-28:
thank you for the help :), this might work for the time being

however I feel like this should be part of the `BufferedUart` API that it just handles the half duplex configuration the same way `Uart` does (using `BufferedUart::new_half_duplex()`), and the user doesn't have to split `tx` and `rx`