embassy-rs/embassy #3298: STM32F1: `adc::AdcChannel` and `adc::AnyAdcChacnnel` are Missing

toon23, 2024-08-30:
Regarding to the [documentation](https://docs.embassy.dev/embassy-stm32/git/stm32f103c8/adc/index.html) the struct AnyAdcChannel and the Trait AdcChannel should be accessible.

In my case `use embassy_stm32::adc::{AdcChannel, AnyAdcChannel};` lead to following compile error:
```shell
error[E0432]: unresolved imports `embassy_stm32::adc::AdcChannel`, `embassy_stm32::adc::AnyAdcChannel`
 --> src/main.rs:6:26
  |
6 | use embassy_stm32::adc::{AdcChannel, AnyAdcChannel};
  |                          ^^^^^^^^^^  ^^^^^^^^^^^^^ no `AnyAdcChannel` in `adc`
  |                          |
  |                          no `AdcChannel` in `adc
```

From my Cargo.toml:
```toml
[dependencies.embassy-stm32]
version = "0.1.0"
features = [
  "memory-x",
  "stm32f103c8",
  "unstable-pac",
  "time-driver-tim2",
  "defmt"
]
```

I'm not sure what I'm missing here.

zUnixorn, 2024-09-11:
I have the same problem with the stm32f411ce, maybe I am missing something. But the documentation suggests, that it should work.

zUnixorn, 2024-09-13:
@toon23 I got it working now. If you are using the crates.io version directly it does not work. You need to be using the github version. At least in my case, that was the fix. The embassy book also shows how to do this: https://embassy.dev/book/#_cargo_toml_2

toon23, 2024-09-14:
Thank you for the hint @zUnixorn.

closed 2024-09-14 by toon23