embassy-rs/embassy #3308: need more than 5 future joins, join_array limited usefulness

Gibbz, 2024-09-04:
I need more than 5 joins in some places.
https://docs.embassy.dev/embassy-futures/git/default/join/index.html

I can't use join_array as they are different functions. But they all don't return any result, so should work. This seems like a strange limitation of the join_array which limits it's usefulness. 

Is there some way to override this behaviour on the join_array?

PegasisForever, 2024-09-07:
Use the `join!` macro from the `futures` crate:

Add this to Cargo.toml:

```toml
futures = { version = "0.3.30", default-features = false, features = [
    "async-await",
] }
```

And import:

```rust
use futures::join;
```

Gibbz, 2024-09-13:
Oh thanks!
I assumed they were different than the embassy futures. I'll give it a shot and report back

Gibbz, 2024-09-15:
That looks to do the job!

closed 2024-09-15 by Gibbz