embassy-rs/embassy #3321: how a generic pin

adavida, 2024-09-08:
currently testing embassy on stm32f401 (blackpill board) 

I want to read the value of adc from a list: 
`
for pin in pins.iter_mut() {
        info!("{}", adc.read(pin));
 }
`
or
`
for analog in pins.iter_mut() {
        info!("{}", analog.read());
 }
`
it was possible to generate list pins ? 

I want to read the value of adc from a list: 

thanks

Dirbaio, 2024-09-08:
You can do [`.degrade_adc()`](https://docs.embassy.dev/embassy-stm32/git/stm32h743zi/adc/trait.AdcChannel.html) on a pin, which converts all pins to the same type, [`AnyAdcChannel`](https://docs.embassy.dev/embassy-stm32/git/stm32h743zi/adc/struct.AnyAdcChannel.html). 

See example: https://github.com/embassy-rs/embassy/blob/main/examples/stm32h7/src/bin/adc_dma.rs#L55-L68

adavida, 2024-09-09:
Thank 
this work ;-)

by only on master

closed 2026-11-06 by leftger