embassy-rs/embassy #3339: Embassy boot update signature verification does not return an error when no signature feature is enabled

kingofpayne, 2024-09-16:
In the `embassy-boot` library, the documentation from the `verify_and_mark_updated` method says:

> If no signature feature is set then this method will always return a signature error.

However, the implementation seems to call `mark_updated` without returning any error when no signature feature is enabled.
I think there is either a mistake in the documentation or in the implementation (the later seems to be the case for me).

Note: the code is available in blocking and async methods, both may need to be patched.

https://github.com/embassy-rs/embassy/blob/3d6a270f30c45eaf394c8eb8bf182dd1a7ec2d7b/embassy-boot/src/firmware_updater/asynch.rs#L85

https://github.com/embassy-rs/embassy/blob/3d6a270f30c45eaf394c8eb8bf182dd1a7ec2d7b/embassy-boot/src/firmware_updater/blocking.rs#L120

lulf, 2024-09-16:
Yup, that seems like bug. Would you like to submit a PR with a fix?

kingofpayne, 2024-09-16:
Thanks for the confirmation. I will give a try to make a PR.

PR #3340 (2024-09-16): fix(boot): return signature error when no features
Fixes #3339
Always return signature error in verify_and_mark_updated when no signature features are enabled.
closed 2024-09-16 by lulf