embassy-rs/embassy #3352: Stopping the executor

ruza-net, 2024-09-19:
Is there a way to kill all tasks, stop the executor, and continue execution in a synchronous way?

I'm asking this because I'm making a bootloader, and the executor seems to interfere with jumping to user application.

Dirbaio, 2024-09-19:
There is not.

It shouldn't interfere with jumping to the application. You should be able to disable all irqs in NVIC, then unpend them, then set VTOR then call `cortex_m::asm::bootload` at any time, even from within an async task.

(You need to do it from thread mode, so using the normal Executor, not InterruptExecutor)

If your bootloader is simple enough, you can also not use any executor at all. Either run futures with `embassy_futures::block_on` or not use async at all.

ruza-net, 2024-09-19:
Thanks for the response! I tried the following:
```rust
async unsafe fn jump_to_app() -> ! {
    info!("jumping to firmware");
    Timer::after_secs(2).await;

    cortex_m::peripheral::NVIC::mask(embassy_stm32::interrupt::ETH);
    cortex_m::peripheral::NVIC::mask(embassy_stm32::interrupt::RNG);
    cortex_m::peripheral::NVIC::unpend(embassy_stm32::interrupt::ETH);
    cortex_m::peripheral::NVIC::unpend(embassy_stm32::interrupt::RNG);

    cortex_m::interrupt::disable();
    let addr = 0x08080200;

    let p = cortex_m::Peripherals::steal();
    p.SCB.vtor.write(addr);

    cortex_m::interrupt::enable();
    cortex_m::asm::bootload(addr as *const u32);
}
```

However, but the user application still doesn't start properly (it should blink a LED, but the LED lights up and stays that way). I suspect this is to do with the network stack (ethernet status LEDs don't reset during/after the jump). When I try this with a toy bootloader (that immediately performs the jump), the application works as expected.

Do you have an idea how to fix this? Should I abandon `async`, or even whole `embassy`?

Dirbaio, 2024-09-19:
This is definitely possible to make work.

Is this stm32? Others have reported issues if you don't mask+unpend the irq for the timer used by the time driver. (if you're using `time-driver-any` it's automatically picked, usually one of the higher numbered timers, you can find out which by checking which timer is missing in `Peripherals`)

I'd recommend iterating over all the irq numbers and doing mask+unpend for every single one of them though. There's also DMA, EXTI irqs that the HAL enables for you.

Dirbaio, 2024-09-19:
which exact stm32 are you using?

ruza-net, 2024-09-20:
I'm using the chip stm32f767vgtx. Thanks for the tip with timer interrupts, I'll try the iterating approach.

ruza-net, 2024-09-20:
The jumping works after _configuring clocks in bootloader and in user application to the same settings_ (scaling factors, HSE mode, and such). I wonder if there's a way to dispose of this need of compatibility? Can I somehow reset the clock so that user application can apply its own settings independently?

Dirbaio, 2024-09-20:
It's explicitly supported to have different rcc configs in the bootloader and the application, if both are using `embassy-stm32`. embassy-stm32 has code in rcc init to switch back to HSI before reconfiguring RCC. https://github.com/embassy-rs/embassy/blob/main/embassy-stm32/src/rcc/f247.rs#L151-L157 (If you find this doesn't work please open an issue!)

I'm not sure if stm32h7xx-hal or the ST C HALs do this. If they don't then this could be the cause of the issue. You can probably workaround it by making the bootloader "deinit" rcc itself before jumping to the application: switch to HSI, disable PLLs, disable HSE. There's no API in embassy-stm32 to do this, you can do it using the PAC directly.

ruza-net, 2024-09-20:
Ok, thanks so much for the help!

closed 2024-09-20 by ruza-net