embassy-rs/embassy #3365: Error when running the rng example on stm32l476rg

Dueesberch, 2024-09-23:
When running example/stm32l4/src/bin/rng.rs I get
`
Error: Failed to attach to RTT

Caused by:
    Error attempting to attach to RTT: RTT control block not found in target memory.
    - Make sure RTT is initialized on the target, AND that there are NO target breakpoints before RTT initialization.
    - For VSCode and probe-rs-debugger users, using `halt_after_reset:true` in your `launch.json` file will prevent RTT
    initialization from happening on time.
    - Depending on the target, sleep modes can interfere with RTT.
`
That error is already described here [Issue 2221](https://github.com/embassy-rs/embassy/issues/2221#issue-2011177204), but adding `config.rcc.clk48_src = Clk48Src::PLL1_Q` ends in `unknown field` for clk48_src and `use of undeclared type 'Clk48Src'`

PegasisForever, 2024-11-11:
Try `config.rcc.mux.clk48sel = Clk48sel::PLL1_Q;`

moellerchristian, 2025-01-19:
Solved the issue. Thanks.