embassy-rs/embassy #3381: Flex.as_af_unchecked does not work as expected.

imfaser, 2024-09-28:
when I use Flex.as_af_unchecked() in my code, There is a strange  problem. The gpio moder will be analog mod rather than alternate.

```rust
impl<'d, T: usart::Instance> UsartPin<T, Tx> {
    pub fn new_tx(pin: impl Peripheral<P = impl TxPin<T>> + 'd, output_type: OutputType) -> Self {
        into_ref!(pin);
        let af_num = pin.af_num();
        debug!("usart pin af num: {}", af_num);
        let mut flex = Flex::new(pin);
        flex.set_as_af_unchecked(af_num, AfType::output(output_type, Speed::VeryHigh));
        UsartPin {
            phantom: PhantomData,
        }
    }
}
```
Because of as_af is a private method so i can only use this way to get a good type hint. 
When I want to use directly method to change moder in this function there are no any change. 
The moder is also analog.
```rust
impl<'d, T: usart::Instance> UsartPin<T, Rx> {
    pub fn new_rx(pin: impl Peripheral<P = impl RxPin<T>> + 'd, pull: Pull) -> Self {
        into_ref!(pin);
        let af_num = pin.af_num();
        debug!("usart pin af num: {}", af_num);
        let mut flex = Flex::new(pin);
        flex.set_as_af_unchecked(af_num, AfType::input(pull));
        critical_section::with(|_| {
            //! no use the mode is also analog
            use embassy_stm32::pac::GPIOA as regs;
            regs.moder().modify(|w| w.set_moder(10, Moder::ALTERNATE));
        });
        UsartPin {
            phantom: PhantomData,
        }
    }
}
```
I use embassy_stm32 hal. The mcu is STM32F407Zgt6. I hope hal can add more method to custom the driver.

When I put the code in main function or anywhere else，it can work properly.

```rust
let tx = UsartPin::new_tx(p.PA9, OutputType::PushPull);
    let rx = UsartPin::new_rx(p.PA10, Pull::None);
    // this code is work
    {
        use embassy_stm32::pac::GPIOA as regs;
        regs.moder().modify(|w| w.set_moder(9, Moder::ALTERNATE));
        regs.moder().modify(|w| w.set_moder(10, Moder::ALTERNATE));
  }
```

Dirbaio, 2024-09-29:
Flex restores the pin to Analog when dropped. If you want it to stay in AF mode, store it as a field in `UsartPin`, this way it stays in AF mode until you drop the UsartPin.

imfaser, 2024-09-30:
> Flex restores the pin to Analog when dropped. If you want it to stay in AF mode, store it as a field in `UsartPin`, this way it stays in AF mode until you drop the UsartPin.

The code run normally, thank you

closed 2024-09-30 by imfaser