embassy-rs/embassy #3398: Fails to enable VREF on STM32G0

drbartling, 2024-10-07:
I'm unable to turn on VREF using embassy on an STM32G0 part.  I've included an example project for reference that builds and demonstrates the issue.  I'm not sure what I'm doing wrong in this case, since looking through the stack, it seems that the procedure very closely resembles the C example.

References:
https://github.com/drbartling/vref-question

I would expect that if there are caps on the output of vref+, that I would be able to measure ~2.5V on the pin, and I should get a credible result from a vref measurement.

IvanLi-CN, 2024-11-23:
I think you should do something like this:

```rust
const VREFBUF_CSR_ADDR: *mut u16 = 0x40010030 as *mut u16;

async fn main() {
    unsafe {
        write_volatile(VREFBUF_CSR_ADDR, 0x0000_0005 as u32); // 2.5v and output from Vref+ pin
    }
}
```

Refer to section 16.2 of RM0444. The code above has been adjusted for the G4 I'm using, and it should be correct. Perhaps after resolving [this issue](https://github.com/embassy-rs/stm32-data/issues/305), there will be a decent configuration method available.

leftger, 2026-11-06:
Fixed by #7438 (commit 5bde61d).

closed 2026-11-06 by leftger