embassy-rs/embassy #3399: embassy_stm32: How to send a large amount of the same data using SPI?

jmjoy, 2024-10-07:
I am using an SPI LCD screen, one of the operations is to full the screen, which is actually writing the same color data as the number of pixels on the screen.

Using the SPL library, the SPI DMA channel can be set up like this:

```c
DMA_InitTypeDef DMA_InitStructure;
DMA_InitStructure.DMA_DIR = DMA_DIR_PeripheralDST;
DMA_InitStructure.DMA_MemoryInc = DMA_MemoryInc_Disable;
DMA_InitStructure.DMA_M2M = DMA_M2M_Disable;
// ...
DMA_Init(DMA_CHx, &DMA_InitStructure);
```

Because the memory address does not increase, the specified amount of the same data will be written to MOSI.

But I can't find a similar method in `embassy_stm32::spi::Spi`. It can be done using `write`, but if the screen pixels are large, the stack will explode, and using the heap is not advisable.

I hope there will be something like:

```rust
async fn write_n(&mut self, data: Word, n: usize) -> Result<(), Self::Error>;
```

showier-drastic, 2024-10-07:
You can open a relatively small array (such as 32bytes), and send it repeatedly.

jmjoy, 2024-10-08:
> You can open a relatively small array (such as 32bytes), and send it repeatedly.

This is a good solution for the current situation.