embassy-rs/embassy #3403: Build Error: no method named `as_raw` found for reference `&Waker` in the c  urrent scope

omani, 2024-10-08:
```
error[E0599]: no method named `as_raw` found for reference `&Waker` in the c
 urrent scope
   --> /home/sun/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-ex
 ecutor-0.6.0/src/raw/waker.rs:53:35
    |
 53 |             let raw_waker = waker.as_raw();
    |                                   ^^^^^^
    |
 help: there is a method `as_ref` with a similar name
    |
 53 |             let raw_waker = waker.as_ref();
    |                                   ~~~~~~

 For more information about this error, try `rustc --explain E0599`.
 error: could not compile `embassy-executor` (lib) due to 1 previous error
```

seems to come from this PR #3059

Dirbaio, 2024-10-08:
This happens because you're using a too old nightly, try updating it.

omani, 2024-10-08:
what does "too old" mean in this context? a few days?
```
$ cargo version
cargo 1.83.0-nightly (ad074abe3 2024-10-04)
$ rustc --version
rustc 1.83.0-nightly (9096f4faf 2024-10-05)
```

closed 2024-10-08 by omani