embassy-rs/embassy #3413: Missing rtr flag setup on embassy_stm32

shestakovvv, 2024-10-11:
embassy/embassy-stm32/src/can/bxcan/registers.rs:
    
```
...
    fn write_mailbox(&self, idx: usize, frame: &Frame) {
        debug_assert!(idx < 3);

        let mb = self.0.tx(idx);
        mb.tdtr().write(|w| w.set_dlc(frame.header().len() as u8));

        mb.tdlr()
            .write(|w| w.0 = u32::from_ne_bytes(unwrap!(frame.data()[0..4].try_into())));
        mb.tdhr()
            .write(|w| w.0 = u32::from_ne_bytes(unwrap!(frame.data()[4..8].try_into())));
        let id: IdReg = frame.id().into();
        mb.tir().write(|w| {
            w.0 = id.0;
            w.set_txrq(true);
            // if frame.header().rtr() {
            //    w.set_rtr(Rtr::REMOTE);
            // }
        });
    }
...
```

Dirbaio, 2024-10-13:
can you send a PR?

TuEmb, 2024-10-16:
@Dirbaio I created a PR for it at https://github.com/embassy-rs/embassy/pull/3421

plaes, 2024-10-24:
@Dirbaio @TuEmb This could be closed?

TuEmb, 2024-10-24:
I think we should close this issue

closed 2024-10-24 by Dirbaio