embassy-rs/embassy #3438: no method named `vtable` found for reference `&Waker` in the current scope

REN-aissance, 2024-10-21:
Hello, I was just now working on a project and was encountering some issues so I tried cargo updating. Now I am receiving this error and I cannot downgrade to embassy-executor 0.6.0 since it was yanked from crates.io. Am I missing something?

no method named `vtable` found for reference `&Waker` in the current scope
no method named `data` found for reference `&Waker` in the current scope
~/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-executor-0.6.1/src/raw/waker.rs:53:36

REN-aissance, 2024-10-21:
My bad, I found the actual embassy-executor merge and saw the change to the working nightly version (2024-09-06 for anyone reading this after the fact and is similarly confused)

closed 2024-10-21 by REN-aissance

youpie, 2025-05-01:
@REN-aissance How did you manage to fix this issue?