embassy-rs/embassy #3448: HAL conflict

Erhannis, 2024-10-23:
I'm trying to add Embassy to my (existing) RP2040 project.  However, I get linker errors unless I also remove rp2040-hal.  While I could probably live with the annoyance of having to rewrite the half of my code that depended on that, one of the lines that now error is `gpio0.set_output_override(rp2040_hal::gpio::OutputOverride::Invert);`, and this doesn't appear to have an embassy-rp equivalent.  I can only assume there are other similar discrepancies.  Can Embassy coexist with another HAL?

lulf, 2024-10-23:
You can use embassy with any HAL you like, but using both embassy-rp HAL and rp2040-hal HAL is a bit risky you can't guarantee they don't step on each others toes.

Could you provide the errors you are seeing?

Erhannis, 2024-10-23:
Oh - I thought `embassy-rp` was _how you used_ embassy on an rp2040.  All the documentation and examples I've seen appear to include an embassy HAL.  How do you use embassy without one?

lulf, 2024-10-23:
Well, the HAL is just for using peripherals of your chip. Embassy-rp is a HAL that supports async rust out of the box, which i dont think is the case for rp2040-hal. 

So if you want to use embassy executor with rp2040-hal, its possible. But since that HAL does not support async, you get little benefit from using embassy executor  in the first place.

lulf, 2024-10-23:
So to use embassy with rp2040-hal, you basically create the executor as you would with any hal:

```
#[embassy_executor::main]
async fn main(s: Spawner) {
    // Use rp2040-hal to initialize and use peripherals
}
``` 

I don't have any more detailed examples, but if you combine examples from rp2040-hal into the above main fn, that's a start.

Erhannis, 2024-10-24:
I got things to compile, then had problems with "Failed to attach to RTT, continuing" and my code not actually running, but I think I figured it out.  While I'm not sure I've figured out all the weirdness that was going on, I did get a program running.  Ironically, the non-running code related to a behavior I noticed like 3 months ago and asked about on Stack Overflow, and then forgot.  https://stackoverflow.com/questions/78732542/rust-rp2040-entry-differences

I think at several points I removed the `rp_pico` crate, not needing it, and just using `rp2040_hal`.  However, as it turns out, `rp_pico` declares the bootloader.  If I didn't `use rp_pico` in some way, the bootloader didn't get written, and the code didn't run.  So adding this to the main rs file fixed it:
```
#[link_section = ".boot2"]
#[no_mangle]
#[used]
pub static BOOT2_FIRMWARE: [u8; 256] = rp2040_boot2::BOOT_LOADER_W25Q080;
```

I'm not sure what was going on with the linking errors, but unless they happen again I probably won't investigate further.

You may be right about the "little benefit" thing - seems like embassy-time required a platform dependent implementation, provided by embassy-rp, so I'm not sure I'll even be able to have delays; guess we'll see how that shakes out.  Is there a way to voluntarily yield to other tasks without anything specific to wait for, so I could e.g. busy-poll in several tasks at once?

closed 2024-10-24 by Erhannis

lulf, 2024-10-25:
@Erhannis You can use https://docs.rs/embassy-futures/0.1.1/embassy_futures/fn.yield_now.html