embassy-rs/embassy #3450: no method named `_` found for reference `&Waker` in the current scope

rowanfr, 2024-10-23:
When trying to compile using dependency `embassy-executor = { version = "0.6.1", features = ["nightly", "integrated-timers"] }` on the `esp` toolchain installed by `espup` I get compiler error:
```rust
error[E0599]: no method named `vtable` found for reference `&Waker` in the current scope
  --> /Users/rowanfr/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-executor-0.6.1/src/raw/waker.rs:53:20
   |
53 |             (waker.vtable(), waker.data())
   |                    ^^^^^^ method not found in `&Waker`

error[E0599]: no method named `data` found for reference `&Waker` in the current scope
  --> /Users/rowanfr/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-executor-0.6.1/src/raw/waker.rs:53:36
   |
53 |             (waker.vtable(), waker.data())
   |                                    ^^^^ method not found in `&Waker`
```

This is due to my toolchain being nightly and the current accessors not being available for type `Waker`. The relevant code snippets are:
https://github.com/embassy-rs/embassy/blob/8803128707b8bd9fc9dcea392a62dfd42aa822d2/embassy-executor/src/raw/waker.rs#L51-L54

```
pub struct Waker {
    waker: RawWaker,
}
```

```
pub struct RawWaker {
    /// A data pointer, which can be used to store arbitrary data as required
    /// by the executor. This could be e.g. a type-erased pointer to an `Arc`
    /// that is associated with the task.
    /// The value of this field gets passed to all functions that are part of
    /// the vtable as the first parameter.
    data: *const (),
    /// Virtual function pointer table that customizes the behavior of this waker.
    vtable: &'static RawWakerVTable,
}
```

I presume that this section in the executor should be altered to use `RawWaker` but there is a chance that the `esp` toolchain is at fault and nothing needs to be done in this code base. I will look further into this but I'm not familiar with `Waker`.

decaday, 2024-10-27:
This is because of the nightly API changes.
waker_getters have been stabilized for 1.83. See https://github.com/rust-lang/rust/issues/96992.

https://github.com/embassy-rs/embassy/commit/01d8508b6c9bf89e27a83b9587bcc13c358c1e51

rowanfr, 2024-10-28:
Closing the issue as @decaday correctly pointed out the API change and thus this has nothing to do with embassy

closed 2024-10-28 by rowanfr