embassy-rs/embassy #3459: stm32 hid Control Pipe data out is unstable

votrungchi, 2024-10-25:
Hello, I am facing a problem with USB HID on STM32F407VG, the data out from Control Pipe is incorrect. The data I only send is [Tx Feature Report [2] --> 0102], but the Control Pipe data out sometime is [00, 00], sometime is [ ], the expected is [01, 02].

The code I use to reproduce the issue.

```
#![no_std]
#![no_main]

use {defmt_rtt as _, panic_probe as _};

use embassy_stm32::usb::Driver;
use embassy_stm32::{bind_interrupts, usb, Config};
use embassy_stm32::time::Hertz;

use embassy_executor::Spawner;
use embassy_stm32::peripherals::USB_OTG_FS;
use embassy_usb::class::hid::{HidWriter, State};
use embassy_usb::Builder;

use my_crate::handler::MyDeviceHandler;

#[rustfmt::skip]
pub const MY_TEST_DESCRIPTOR: [u8; 37] =
        [
            0x05, 0x01,              // USAGE_PAGE (Generic Desktop)
            0x09, 0x00,              // USAGE (Undefined)
            0xA1, 0x01,              // COLLECTION (Application)

            // Feature Report
            0x85, 0x01,          // REPORT_ID (1)
            0x09, 0x02,          // USAGE (Undefined)
            0x15, 0x00,          // LOGICAL_MINIMUM (0)
            0x26, 0xFF, 0x00,    // LOGICAL_MAXIMUM (255)
            0x75, 0x08,          // REPORT_SIZE (8)
            0x95, 0x01,          // REPORT_COUNT (1)
            0xB1, 0x02,          // FEATURE (Data,Var,Abs)

            // Output Report
            0x85, 0x02,          // REPORT_ID (2)
            0x09, 0x03,          // USAGE (Undefined)
            0x15, 0x00,          // LOGICAL_MINIMUM (0)
            0x26, 0xFF, 0x00,    // LOGICAL_MAXIMUM (255)
            0x75, 0x08,          // REPORT_SIZE (8)
            0x95, 0x01,          // REPORT_COUNT (1)
            0x91, 0x02,          // OUTPUT (Data,Var,Abs)

            0xC0
        ];

bind_interrupts!(struct Irqs {
    OTG_FS => usb::InterruptHandler<USB_OTG_FS>;
});

#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    let mut config = Config::default();
    {
        use embassy_stm32::rcc::*;
        config.rcc.hse = Some(Hse {
            freq: Hertz(8_000_000),
            mode: HseMode::Bypass,
        });
        config.rcc.pll_src = PllSource::HSE;
        config.rcc.pll = Some(Pll {
            prediv: PllPreDiv::DIV4,
            mul: PllMul::MUL168,
            divp: Some(PllPDiv::DIV2), // 8mhz / 4 * 168 / 2 = 168Mhz.
            divq: Some(PllQDiv::DIV7), // 8mhz / 4 * 168 / 7 = 48Mhz.
            divr: None,
        });
[... 107 lines cut ...]
TRACE irq
└─ embassy_stm32::usb::_version::{impl#0}::on_interrupt @ embassy\embassy-stm32\src\usb\otg.rs:27
TRACE irq
└─ embassy_stm32::usb::_version::{impl#0}::on_interrupt @ embassy\embassy-stm32\src\usb\otg.rs:27
TRACE control request: Request { direction: Out, request_type: Class, recipient: Interface, request: 9, value: 769, index: 0, length: 2 }
└─ embassy_usb::{impl#1}::handle_control::{async_fn#0} @ embassy\embassy-usb\src\lib.rs:347
TRACE irq
└─ embassy_stm32::usb::_version::{impl#0}::on_interrupt @ embassy\embassy-stm32\src\usb\otg.rs:27
TRACE   control out data: [01, 02]
└─ embassy_usb::{impl#1}::handle_control_out::{async_fn#0} @ embassy\embassy-usb\src\lib.rs:425
TRACE HID control_out Request { direction: Out, request_type: Class, recipient: Interface, request: 9, value: 769, index: 0, length: 2 } [1, 2]
└─ embassy_usb::class::hid::{impl#8}::control_out @ embassy\embassy-usb\src\class\hid.rs:468
TRACE irq
└─ embassy_stm32::usb::_version::{impl#0}::on_interrupt @ embassy\embassy-stm32\src\usb\otg.rs:27
TRACE irq
└─ embassy_stm32::usb::_version::{impl#0}::on_interrupt @ embassy\embassy-stm32\src\usb\otg.rs:27
TRACE control request: Request { direction: Out, request_type: Class, recipient: Interface, request: 9, value: 769, index: 0, length: 2 }
└─ embassy_usb::{impl#1}::handle_control::{async_fn#0} @ embassy\embassy-usb\src\lib.rs:347
TRACE irq
└─ embassy_stm32::usb::_version::{impl#0}::on_interrupt @ embassy\embassy-stm32\src\usb\otg.rs:27
TRACE   control out data: [01, 02]
└─ embassy_usb::{impl#1}::handle_control_out::{async_fn#0} @ embassy\embassy-usb\src\lib.rs:425
TRACE HID control_out Request { direction: Out, request_type: Class, recipient: Interface, request: 9, value: 769, index: 0, length: 2 } [1, 2]
└─ embassy_usb::class::hid::{impl#8}::control_out @ embassy\embassy-usb\src\class\hid.rs:468
```

Dirbaio, 2024-10-25:
Can you try enabling trace logs in embassy-usb-synopsys-otg? Add it here https://github.com/embassy-rs/embassy/blob/main/embassy-stm32/Cargo.toml#L113 so the `defmt` feature of `embassy-stm32` enables it, then make sure you have `DEFMT_LOG=trace`

votrungchi, 2024-10-26:
Hello, here is the trace: [hid.log](https://github.com/user-attachments/files/17528628/hid.log)
Note: I have to comment out the assert because of compile error: 
```
Compiling embassy-usb-synopsys-otg v0.1.0 (C:\Users\Chi\workspace\github.com\votrungchi\formula-rs\embassy\embassy-usb-synopsys-otg)
error[E0015]: cannot call non-const fn `defmt::export::acquire` in constant functions
   --> embassy\embassy-usb-synopsys-otg\src\fmt.rs:16:13
    |
16  |             ::defmt::assert!($($x)*);
    |             ^^^^^^^^^^^^^^^^^^^^^^^^
```
```
    #[doc = "Device IN endpoint transmit FIFO size register"]
    #[inline(always)]
    pub const fn dieptxf(self, n: usize) -> Reg<regs::Fsiz, RW> {
        // assert!(n < 7usize);
        unsafe { Reg::from_ptr(self.ptr.add(0x0104usize + n * 4usize) as _) }
    }
```

votrungchi, 2024-10-29:
This error also happen after the data is wrong:
```
TRACE control request: Request { direction: Out, request_type: Class, recipient: Interface, request: 9, value: 772, index: 0, length: 4 }
└─ embassy_usb::{impl#1}::handle_control::{async_fn#0} @ embassy\embassy-usb\src\lib.rs:347
ERROR panicked at C:\Users\Chi\workspace\github.com\votrungchi\formula-rs\embassy\embassy-usb-synopsys-otg\src\lib.rs:46:9:
assertion failed: ep_num < ep_count
└─ panic_probe::print_defmt::print @ C:\Users\Chi\.cargo\registry\src\index.crates.io-6f17d22bba15001f\panic-probe-0.3.2\src\lib.rs:104
```

commit 36c7891b (2024-11-22): otg: fix corruption in CONTROL OUT transfers in stm32f4.

The RM says we have to process STUP (and therefore clear CNAK to start the data stage)
in the DOEPINT STUP interrupt. Seems doing it in RXFLVL when we receive the data is
too early. This makes it work consistently on all chips, so the quirk is no longer needed.

Fixes #3493
Fixes #3459
PR #3565 (2024-11-23): otg: fix corruption in CONTROL OUT transfers in stm32f4.       

The RM says we have to process STUP (and therefore clear CNAK to start the data stage)
in the DOEPINT STUP interrupt. Seems doing it in RXFLVL when we receive the data is
too early. This makes it work consistently on all chips, so the quirk is no longer needed.

Fixes #3493
Fixes #3459

- **otg: fix build with defmt enabled.**
- **otg: improve trace logging, print bytes as hex.**
- **otg: fix corruption in CONTROL OUT transfers in stm32f4.**
commit 72353387 (2024-11-23): otg: fix corruption in CONTROL OUT transfers in stm32f4.        (#3565)

* otg: fix build with defmt enabled.

* otg: improve trace logging, print bytes as hex.

* otg: fix corruption in CONTROL OUT transfers in stm32f4.

The RM says we have to process STUP (and therefore clear CNAK to start the data stage)
in the DOEPINT STUP interrupt. Seems doing it in RXFLVL when we receive the data is
too early. This makes it work consistently on all chips, so the quirk is no longer needed.

Fixes #3493
Fixes #3459
closed 2024-11-23 by Dirbaio