embassy-rs/embassy #3464: RP2040 : Is it possible to write flash at 0x10000000 in runtime (without probe or usb) ?

Athorus, 2024-10-26:
Currently my Pico software is able to update the DFU partition through wifi, which is nice ! I would like to update the boatloader too, but the Pico seems to crash when I try to erase the first 4096 bytes.
Is it possible or do I have to forget it ?

Thanks

Athorus, 2024-10-27:
In embassy_rp/flash.rs we can see the comment below. So I wrote the first sector of flash before launching the second core and any async code and it seems to work.

    /// Nothing must access flash while this is running.
    /// Usually this means:
    ///   - interrupts must be disabled
    ///   - 2nd core must be running code from RAM or ROM with interrupts disabled
    ///   - DMA must not access flash memory
    ///
    /// `addr` and `len` parameters must be valid and are not checked.

Athorus, 2024-11-08:
@Georges760  I don’t see why this PR is related to my issue.

Georges760, 2024-11-08:
it is not, sorry to have tag your issue, it was a typo ;)