embassy-rs/embassy #3465: FR: Method for measuring CPU load

lure23, 2024-10-26:
I'm starting to use Embassy and one curiousity is about how much CPU headroom there is, when I'm running a certain application. In other words, how long does the Executor sleep?

This issue is intended as a place for discussion / collection of knowledge. I start by listing the few resources I've found.

Since I am new to Rust and Embassy, I am not yet prepared for providing a PR. However, I will be experimenting with this need.

- ["Measuring Cpu usage with rust Embassy"](https://www.giacomocaironi.dev/posts/measuring-cpu-usage-with-rust-embassy/) (blog, Jun 2024; on "ARM 32-bit")
- ["Add a (optional) way to measure CPU load on Cortex M executor"](https://github.com/embassy-rs/embassy/pull/2759) (PR; open since Apr 2024; on Cortex-M)

Luctins, 2024-10-30:
I have a similar issue: I'm uncertain if my current performance bottlenecks are in IO or in in the microcontroller core itself.

Also using the executor itself would be a good bet since the information would be most available there. Maybe tracking some form of Id for each async task could be a start?

Gibbz, 2024-11-10:
It would be good to have a profiling tool. Inhave a basic one I've written for using the time how long code takes to execute.
Possibly having something like this all as one crate would be useful.

lure23, 2024-11-11:
I wrote the FR 2 weeks ago. In the mean time, I have realized the feature (as a centralized monitoring feature) may not be as necessary as I originally imagined. My thinking was desktop based, I presume.

The code I work on (as many embedded code, I imagine) works as repeated "frames", or as a state machine. By timing how much time each operation within a "frame" takes, I have gotten a good understanding of where the time is spent - and as a side benefit how much CPU budget remains, until the core wouldn't be able to carry everything out each "frame".

I will keep the FR open, as it was intended to catch discussion on this issue.

lynxD, 2025-05-19:
Since the tasks are async it not as simple as setting a timestamp in each loop iteration. Even checking all await-s would not provide the whole picture as async functions can await on other async functions. IHMO centralized accounting is something that would be very beneficial. It would allow us to just turn it on at the beginning of development and see if some task takes too long chunks of time.

BTW: Somebody opened a pr to add examples based on the blog post mentioned: https://github.com/embassy-rs/embassy/pull/3920