embassy-rs/embassy #3466: RP2040: embassy-usb-logger does not work outside of embassy

jdswensen, 2024-10-27:
If you copy/paste the code from [the RP2040 USB logger example](https://github.com/embassy-rs/embassy/blob/main/examples/rp/src/bin/usb_logger.rs) into a new project, it will not compile.

Here is the cargo manifest I used:

```
[package]
edition = "2021"
name = "usb-logger"
version = "0.1.0"
resolver = "2"

[dependencies]
defmt = "0.3"
defmt-rtt = "0.4"
panic-probe = { version = "0.3", features = ["print-defmt"] }

embedded-hal = "1.0.0"
embedded-hal-async = "1.0.0"
embedded-io = "0.6.1"
embedded-io-async = "0.6.1"
embedded-storage = "0.3.1"

cortex-m = { version = "0.7.6" }
cortex-m-rt = "0.7.3"

embassy-executor = { version = "0.5.0", features = ["task-arena-size-1024", "arch-cortex-m", "executor-thread", "defmt", "integrated-timers", "executor-interrupt"] }
embassy-futures = "0.1.1"
embassy-rp = { version = "0.1.0", features = ["defmt", "unstable-pac", "time-driver", "critical-section-impl"] }
embassy-sync = { version = "0.6.0" }
embassy-time = { version = "0.3.0", features = ["defmt", "defmt-timestamp-uptime"] }
embassy-usb = "0.3.0"

# usbd-hid = "0.8.2"
# usbd-hid-macros = "0.8.2"
embassy-usb-logger = "0.2.0"
# embassy-usb-logger = { git = "https://github.com/embassy-rs/embassy.git" }
log = "0.4.22"
```

and the compile error:

```
error: failed to select a version for `byteorder`.
    ... required by package `usbd-hid-macros v0.6.0`
    ... which satisfies dependency `usbd-hid-macros = "^0.6.0"` of package `usbd-hid v0.7.0`
    ... which satisfies dependency `usbd-hid = "^0.7.0"` of package `embassy-usb v0.2.0`
    ... which satisfies dependency `embassy-usb = "^0.2.0"` of package `embassy-usb-logger v0.2.0`
    ... which satisfies dependency `embassy-usb-logger = "^0.2.0"` of package `usb-logger v0.1.0 (/Users/jake/projects/tests/usb-logger)`
versions that meet the requirements `~1.4` are: 1.4.3, 1.4.2, 1.4.1, 1.4.0

all possible versions conflict with previously selected packages.

  previously selected package `byteorder v1.5.0`
    ... which satisfies dependency `byteorder = "^1.5"` of package `usbd-hid-macros v0.8.1`
    ... which satisfies dependency `usbd-hid-macros = "^0.8.1"` of package `usbd-hid v0.8.1`
    ... which satisfies dependency `usbd-hid = "^0.8.1"` of package `embassy-usb v0.3.0`
    ... which satisfies dependency `embassy-usb = "^0.3.0"` of package `usb-logger v0.1.0 (/Users/jake/projects/tests/usb-logger)`

failed to select a version for `byteorder` which could resolve this conflict
```

I suspect that this is due to some build error in the tagged version. I tried pointing just the `embassy-usb-logger` crate to the upstream but got a compile error about two different versions of a trait (understandable).

```
error[E0277]: the trait bound `Driver<'_, USB>: embassy_usb_driver::Driver<'_>` is not satisfied
   --> src/main.rs:21:5
    |
21  |     embassy_usb_logger::run!(1024, log::LevelFilter::Info, driver);
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |     |
    |     the trait `embassy_usb_driver::Driver<'_>` is not implemented for `Driver<'_, USB>`
    |     required by a bound introduced by this call
    |
help: trait impl with same name found
   --> /Users/jake/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-rp-0.1.0/src/usb.rs:294:1
    |
294 | impl<'d, T: Instance> driver::Driver<'d> for Driver<'d, T> {
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    = note: perhaps two different versions of crate `embassy_usb_driver` are being used?
note: required by a bound in `UsbLogger::<N>::run`
   --> /Users/jake/.cargo/git/checkouts/embassy-9312dcb0ed774b29/b31648f/embassy-usb-logger/src/lib.rs:67:12
    |
65  |     pub async fn run<'d, D>(&'d self, state: &'d mut LoggerState<'d>, driver: D) -> !
    |                  --- required by a bound in this associated function
66  |     where
67  |         D: Driver<'d>,
    |            ^^^^^^^^^^ required by this bound in `UsbLogger::<N>::run`
    = note: this error originates in the macro `embassy_usb_logger::run` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0277]: the trait bound `Driver<'_, USB>: embassy_usb_driver::Driver<'_>` is not satisfied
   --> src/main.rs:21:5
    |
21  |     embassy_usb_logger::run!(1024, log::LevelFilter::Info, driver);
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `embassy_usb_driver::Driver<'_>` is not implemented for `Driver<'_, USB>`
    |
help: trait impl with same name found
   --> /Users/jake/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-rp-0.1.0/src/usb.rs:294:1
    |
294 | impl<'d, T: Instance> driver::Driver<'d> for Driver<'d, T> {
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    = note: perhaps two different versions of crate `embassy_usb_driver` are being used?
note: required by a bound in `UsbLogger::<N>::run`
   --> /Users/jake/.cargo/git/checkouts/embassy-9312dcb0ed774b29/b31648f/embassy-usb-logger/src/lib.rs:67:12
    |
65  |     pub async fn run<'d, D>(&'d self, state: &'d mut LoggerState<'d>, driver: D) -> !
    |                  --- required by a bound in this associated function
66  |     where
67  |         D: Driver<'d>,
    |            ^^^^^^^^^^ required by this bound in `UsbLogger::<N>::run`
    = note: this error originates in the macro `embassy_usb_logger::run` (in Nightly builds, run with -Z macro-backtrace for more info)

For more information about this error, try `rustc --explain E0277`.
error: could not compile `usb-logger` (bin "usb-logger") due to 3 previous errors
```

So I updated all of the embassy crates to point at the upstream and that seemed to work. It's a good enough work around for me to continue but figured you would like to know!

daveismith, 2024-12-17:
I'd like to also confirm I'm seeing this and that it seems like the last published version of `embassy-usb-logger` does not contain support for the `ReceiverHandler` trait, even though the documentation and examples refer to it. A new release of this crate would be highly appreciated.

Dirbaio, 2025-01-15:
embassy-usb v0.4, embassy-usb-logger v0.4 are now released which should've fixed this. (or I can't repro it anymore, at least). Please let me know if not.

closed 2025-01-15 by Dirbaio