embassy-rs/embassy #3471: nrf9160 example "modem_tcp_client.rs" crashes on nrf9151

tpisto, 2024-10-28:
Error:

```
TRACE ipc 2
└─ embassy_net_nrf91::{impl#3}::poll @ /Users/tommi/EmbeddedRustProjects/nrf9151-test1/embassy-net-nrf91/src/lib.rs:334 
ERROR panicked at 'assertion failed: `(left == right)`'
diff < left / right >
<32
>16
└─ embassy_net_nrf91::{impl#3}::poll @ /Users/tommi/EmbeddedRustProjects/nrf9151-test1/embassy-net-nrf91/src/lib.rs:348 

```
By removing line 348 `assert_eq!(rx_data_len, LIST_LEN);` the example works.

With nrf9151 the rx_data_len at line 348 is 32 instead of 16 (which is LIST_LEN).

lulf, 2024-10-29:
In that case, I think we should probably introduce some features on the net-nrf91 crate to configure both the correct pac crate, as well as the appropriate constants. Thanks for testing it out!

Dirbaio, 2024-10-29:
I think there's no guarantee the list size will stay constant in the future. Maybe we should make it a non-constant.

Also I think the list sizes don't have to match in both directions. We can make *our* list size a constant in the app->modem direction, and trust the modem's size for the modem->app direction (or maybe they're the other way around, not sure)

sebgab, 2025-04-09:
I have made a very rough patch to get it working for the nRF9151 in a fork ([https://github.com/sebgab/embassy](https://github.com/sebgab/embassy)).
All I have done is introduce a feature flag and update the constants in the code based on that flag.

I am a little worried about the stability of this fix over time as I am not too familiar with the driver and how it uses these lengths. I would love to make a proper fix sadly don't have time to do it at this moment.

closed 2026-02-09 by lulf