embassy-rs/embassy #3477: New release for embassy-net.

Frostie314159, 2024-10-29:
Hi, I'd like to ask if a new release of `embassy-net` is planned, the current example for a webrequest on the rp pico w no longer compiles with the crates.io release, due to missing trait impls.

kolbma, 2024-10-30:
Well, that's the way development works.  
It is not possible to release every time there is an API change to get development code to run with released libraries again. ;-)

The cyw43 driver has also changes in its API.

I think you can use  
https://github.com/embassy-rs/embassy/blob/891c5ee10584cd990dad529e3506fe1328e4e69d/examples/rp/src/bin/wifi_webrequest.rs  
with the released crates.

Frostie314159, 2024-10-30:
I'm aware that it's not always possible to release a new version on every change, but I don't see anything blocking a release, especially after a breaking API change.
My use case is actually not related to RP Pico W, because I'm building a Wi-Fi driver for the ESP32, and hence do not depend on cyw43.

AnthonyGrondin, 2024-10-31:
A new release of `smoltcp` is needed first because currently `embassy-net` uses a patch version for core ip types.
https://github.com/embassy-rs/embassy/blob/b0c7fa07b27be9025c5e986e339899cdfc9a01e2/embassy-net/Cargo.toml#L71-L74

I'd also like to implement `writeable`/`readable` methods to `TcpReader`/`TcpWriter` and to UDP sockets before a new release is made.

Dirbaio, 2024-11-28:
embassy-net v0.5.0 released.

closed 2024-11-28 by Dirbaio