embassy-rs/embassy #3479: Breaking type guarantees using conflicting packages.

Arslan8, 2024-10-30:
I am reporting a possible mechanism to break the isolation guarantees based on the unsafe code restriction. Currently, the OS depends on the Embassy HAL (based on stm32 PAC) to manage isolation between devices. Since we cannot use unsafe code, there is no way for different tasks to steal a device inside a task, guaranteeing isolation between devices. 

However, a quick glance at dependents of the HAL crate shows other crates may use the HAL as well. Fortunately, the device ownership variable is unmangled, maintaining a single version of ownership across different versions of the HAL. However, the dependent crates are not under the same scrutiny as a task, and similar to this OS, it steals the device to avoid panic. A task is free to use these conflicting packages to bypass the restrictions enforced by the OS and HAL. Unfortunately, the compiler can only check unsafe usage at a crate level. Therefore, if the app uses any of the dependant packages it can use a safe interface to break the isolation guarantees of device-level isolation. 

I am attaching an [application](https://github.com/user-attachments/files/17577683/rusty-hal.zip) that gives a PoC for this exploit. The exploit is based on RTIC OS, but the same concern applies to Embassy and Embassy HAL as well. To mitigate this issue, I propose maintaining a list of blocklisted/banned crates so that users cannot break the isolation guarantees. We can use [cargo-deny](https://github.com/EmbarkStudios/cargo-deny) to ensure this is satisfied during compilation.

lulf, 2024-10-31:
What isolation guarantees? In bare metal rust, not using an MPU, there is not really any way to prevent any crate or your own code for that matter, to not step on the toes of the HAL.

Arslan8, 2024-10-31:
I see.. so embassy doesn't mandate only safe code should be used in tasks/futures. We can close this issue in that case. Thanks for the clarification.

closed 2024-10-31 by Arslan8