embassy-rs/embassy #3493: panic! in embassy-usb-synopsys-otg on second serial connection (BUT only if defmt-trace level debugging not enabled)

paulc, 2024-11-02:
I had a strange issue running the `usb_serial` example on an STM32F401 where the serial device would work fine for the first connection from the host, but if I disconnected the terminal session and then re-connected the driver would panic with the following message.

```
95.600067 [panic_probe::print_defmt::print] lib.rs:104  ERROR panicked at /Users/paulc/Sync/Development/embedded-rust/embassy/embassy-usb-synopsys-otg/src/lib.rs:46:9:
assertion failed: ep_num < ep_count
```

Very strangely, if I enabled the `defmt` feature in embassy-stm32 crate and set DEFMT_LOG=trace the problem would go away. As an aside, enabling the `defmt` feature in the embassy-usb-synopsys-otg results in lots of compile errors with `defmt::assert!` (see #3492) (constant function errors) which makes debugging a bit tricky.

It looks like the driver is getting an unexpected response from [let status = r.grxstsp().read()](https://github.com/embassy-rs/embassy/blob/fcbbef01cd3c5292be29b78b674f0593277545e7/embassy-usb-synopsys-otg/src/lib.rs#L41) (status.epnum() was 12 rather than 1 when the error showed up in my testing).

The patch below fixes this by replacing the [assert!(ep_num < ep_count)](https://github.com/embassy-rs/embassy/blob/fcbbef01cd3c5292be29b78b674f0593277545e7/embassy-usb-synopsys-otg/src/lib.rs#L46C17-L46C23) with a test which skips the packet rather than panic-ing.

I suspect that there is a more fundamental issue here but but this does at least fix the panic! issue.

```
diff --git a/embassy-usb-synopsys-otg/src/lib.rs b/embassy-usb-synopsys-otg/src/lib.rs
index f90403936..3197817ee 100644
--- a/embassy-usb-synopsys-otg/src/lib.rs
+++ b/embassy-usb-synopsys-otg/src/lib.rs
@@ -43,7 +43,16 @@ pub unsafe fn on_interrupt<const MAX_EP_COUNT: usize>(
         let ep_num = status.epnum() as usize;
         let len = status.bcnt() as usize;
 
-        assert!(ep_num < ep_count);
+        if ep_num >= ep_count {
+            // Removed `assert!(ep_num < ep_count);` as this fails for second serial connection UNLESS
+            // defmt-trace level debugging is enabled in embassy-stm32 (note that the defmt feature
+            // doesnt work in embassy-usb-synopsys-otg as defmt::assert! generates multiple errors)
+            //
+            // I'm guessing this may be some sort of timing issue so instead of panic-ing skip the
+            // invalid packet :‑(
+            error!("Skipping Invalid Packet (ep_num >= ep_count): ep_num: {} ep_count: {}", ep_num, ep_count);
+            continue;
+        }
 
         match status.pktstsd() {
             vals::Pktstsd::SETUP_DATA_RX => {
```

vDorst, 2024-11-03:
I see these panics to with this https://github.com/embassy-rs/embassy/pull/3212#issuecomment-2433589643 usb audio. But every reset, pressing the reset butten, the device/software behaves differently.

votrungchi, 2024-11-04:
I think it related my issue https://github.com/embassy-rs/embassy/issues/3459

elagil, 2024-11-13:
I have started to look into this issue, but have not found the problem yet.

> I see these panics to with this [#3212 (comment)](https://github.com/embassy-rs/embassy/pull/3212#issuecomment-2433589643) usb audio. But every reset, pressing the reset butten, the device/software behaves differently.

Currently, I am looking at https://github.com/ChibiOS/ChibiOS/blob/master/os/hal/ports/STM32/LLD/OTGv1/hal_usb_lld.c as a reference, which is a well tested C implementation. It works reliably for something like USB audio - I made a USB audio class in C first (with that ChibiOS driver) and had no issues on a STM32F401.

votrungchi, 2024-11-14:
The control pipe-out data did not receive the correct data in the first place; this also happened with the hid_raw example of stm32f4.
I feel that the interruption has not been managed well.
Is it the right behavior if the SETUP_DATA_DONE happens before the device sends the required control data (descriptor) to the host?

commit 36c7891b (2024-11-22): otg: fix corruption in CONTROL OUT transfers in stm32f4.

The RM says we have to process STUP (and therefore clear CNAK to start the data stage)
in the DOEPINT STUP interrupt. Seems doing it in RXFLVL when we receive the data is
too early. This makes it work consistently on all chips, so the quirk is no longer needed.

Fixes #3493
Fixes #3459
PR #3565 (2024-11-23): otg: fix corruption in CONTROL OUT transfers in stm32f4.       

The RM says we have to process STUP (and therefore clear CNAK to start the data stage)
in the DOEPINT STUP interrupt. Seems doing it in RXFLVL when we receive the data is
too early. This makes it work consistently on all chips, so the quirk is no longer needed.

Fixes #3493
Fixes #3459

- **otg: fix build with defmt enabled.**
- **otg: improve trace logging, print bytes as hex.**
- **otg: fix corruption in CONTROL OUT transfers in stm32f4.**
commit 72353387 (2024-11-23): otg: fix corruption in CONTROL OUT transfers in stm32f4.        (#3565)

* otg: fix build with defmt enabled.

* otg: improve trace logging, print bytes as hex.

* otg: fix corruption in CONTROL OUT transfers in stm32f4.

The RM says we have to process STUP (and therefore clear CNAK to start the data stage)
in the DOEPINT STUP interrupt. Seems doing it in RXFLVL when we receive the data is
too early. This makes it work consistently on all chips, so the quirk is no longer needed.

Fixes #3493
Fixes #3459
closed 2024-11-23 by Dirbaio