embassy-rs/embassy #3497: STM32U5 LPTIM could be time driver?

chrenderle, 2024-11-03:
[Here](https://github.com/embassy-rs/embassy/blob/c75e1e2a9ed58fbd6c4d08426a26d1505957c68b/embassy-stm32/src/time_driver.rs#L21C1-L30C62) the requirements for the time drivers are described. At least 2 CC registers are required. The STM32U5-Series has low-power timers with 2 CC registers.
[image]

From my understanding it would be possible to use the LPTIMs as time drivers. 
Am I correct here?

This would enable the use of the low-power feature without the RTC.

trembel, 2024-11-04:
I guess that's what I (everyone?) would prefer:

> Ultimately I think the better fix will be to make a time driver using LPTIM, instead of using RTC. It has a few advantages, like not needing to dynamically switching between high-frequency TIMx and low-frequency RTC, which causes losing ticks while switching. The current low-power support uses RTC because the person who contributed it was using an F4 which doesn't have LPTIM.

_Originally posted by @Dirbaio in https://github.com/embassy-rs/embassy/issues/3213#issuecomment-2286681199_

chrenderle, 2024-11-04:
The question is how you would implement the lptim time driver on the mcus that only have 1 CC register. The only series I know which has more than 1 CC for the lptim is the STM32U5-series.

leftger, 2026-11-05:
Embassy now supports LPTIM as a time driver.

closed 2026-11-05 by leftger