embassy-rs/embassy #3499: STM32U5: Wake-Up event after ~30s sleep

trembel, 2024-11-04:
Currently testing the new low-power feature for the STM32U5(45RETxQ) (#3496).

When going directly to sleep (without a time set), the device wakes-up after ~30s of sleep (probably due to Overflow IRQ):
[image]

Code to reproduce:
```
#![no_std]
#![no_main]

use core::arch::asm;
use defmt::*;
use embassy_executor::Spawner;
use embassy_stm32::gpio::{AnyPin, Level, Output, Speed};
use embassy_stm32::low_power::Executor;
use embassy_stm32::rcc::LsConfig;
use embassy_stm32::rtc::{Rtc, RtcConfig};
use embassy_stm32::Config;
use embassy_time::Timer;
use static_cell::StaticCell;
use {defmt_rtt as _, panic_probe as _};

use defmt::info;
use {defmt_rtt as _, panic_probe as _};

#[cortex_m_rt::entry]
fn main() -> ! {
    Executor::take().run(|spawner| {
        unwrap!(spawner.spawn(async_main(spawner)));
    })
}

#[embassy_executor::task]
async fn async_main(spawner: Spawner) {
    // Get default configs, adjust clock tree to use external crystal and run at 80Mhz
    let mut p_config = embassy_stm32::Config::default();
    p_config.enable_debug_during_sleep = false;

    let p = embassy_stm32::init(p_config);
    let rtc = Rtc::new(p.RTC, RtcConfig::default());
    static RTC: StaticCell<Rtc> = StaticCell::new();
    let rtc = RTC.init(rtc);
    embassy_stm32::low_power::stop_with_rtc(rtc);

    info!("RCC initialized");
    //unwrap!(spawner.spawn(blinky(p.PA5.into())));
}
```

Two more questions:
- What was the reason to set `MIN_STOP_PAUSE` to 250ms? To me this seems waaay longer than needed?
- Wouldn't it be possible to disable the RTC completely, when there is no timer IRQ / task set for wakeup?

trembel, 2024-11-04:
I guess the issue lies here:
```
ERROR panicked at /.../src/rtc/low_power.rs:55:31:
attempt to multiply with overflow
└─ panic_probe::print_defmt::print @ /.../panic-probe-0.3.2/src/lib.rs:104
```

trembel, 2024-11-04:
With:
- self_ticks: 15304
- other_ticks: 6916
- rtc_ticks: 8388
- psc: 255
- TICK_HZ: 1000000

Obviously, 1_000_000 * 8_388 = 8_388_000_000 is an u32 overflow.

chrenderle, 2024-11-05:
>     * Wouldn't it be possible to disable the RTC completely, when there is no timer IRQ / task set for wakeup?

Right now the RTC is required for the time driver as far as I understand.

I'm currently not able to reproduce your error. Could you please share your Cargo.toml with me. Especially the tick-hz- feature of embassy-time. The only thing I didn't do exactly as in your example is the timeout task. I'm keeping that in because it is always a hassle to flash the MCU when it is always going into Stop2.

trembel, 2024-11-07:
> Right now the RTC is required for the time driver as far as I understand.
>
> I'm currently not able to reproduce your error. Could you please share your Cargo.toml with me. Especially the tick-hz- [...]

Thanks for trying. Here is my Cargo.toml, with embassy being on commit `3d8e987b`:
```
[package]
edition = "2021"
name = "u545_lp_test_rs"
version = "0.0.1"
authors = ["..."]

[dependencies]
defmt = "0.3"
defmt-rtt = "0.4"
panic-probe = { version = "0.3", features = ["print-defmt"] }

embedded-hal = "1.0.0"
embedded-hal-async = "1.0.0"

cortex-m-rt = "0.7.3"

embassy-executor = { version = "0.6.0", features = ["task-arena-size-16384", "arch-cortex-m", "executor-thread", "defmt", "integrated-timers", "executor-interrupt"] }
embassy-sync = { version = "0.6.0" }
embassy-time = { version = "0.3.2", features = ["defmt", "defmt-timestamp-uptime"] }

cortex-m = { version = "0.7.7", features = ["critical-section-single-core"] }
embassy-stm32 = { version = "0.1.0", features = ["defmt", "stm32u545re", "unstable-pac", "memory-x", "time-driver-any", "low-power"]  }

embassy-embedded-hal = { version = "0.2.0", features = ["defmt"] }
embassy-futures = "0.1.1"
static_cell = "2.1.0"

[patch.crates-io]
embassy-stm32 = { path = "../embassy/embassy-stm32" }
embassy-executor = { path = "../embassy/embassy-executor" }
embassy-sync = { path = "../embassy/embassy-sync" }
embassy-time = { path = "../embassy/embassy-time" }
embassy-futures = { path = "../embassy/embassy-futures" }
```

chrenderle, 2024-11-07:
Sadly I'm still not able to reproduce your error. Basically the only difference I have to you is that I have a STM32U575ZI instead of your STM32U545RE. When I order the next time from STM I will get the Nucleo U545-RE as well. 

[image]
As you can see here the MCU wakes up after 32s. This is normal behavior. The RTC needs to wakeup after max 32s to keep the time for the time driver. There is a possibility to implement a longer sleep time from the RTC but it is not yet implemented. And this probably wouldn't fix your issue.