embassy-rs/embassy #3509: Ethernet SPI example not working on RP2350 at 50 MHz

romainreignier, 2024-11-06:
I have tried the RP2040 [`ethernet_w5500_tcp_server`](https://github.com/embassy-rs/embassy/blob/4136892f3f310ac9bdc569692f12a6beafe1475f/examples/rp/src/bin/ethernet_w5500_tcp_server.rs) example on the RP2350 (exact code below) (Pico2 board with external W5500) and it was working at the end of September and it now fails at runtime with this error:

```
0.102403 ERROR invalid chip version: 2 (expected 4)
└─ embassy_net_wiznet::device::{impl#3}::new::{async_fn#0} @ /home/rre/rust/embassy/embassy-net-wiznet/src/device.rs:91
``` 

I have done a git bisect and found that the regression is caused by [this commit](https://github.com/embassy-rs/embassy/commit/b73b3f2da0940907dfc39bec62ff33c976874017) changing the clock frequency to 150 MHz.

I have tried to use logic analyzer to check the SPI clock frequency but I only own a cheap FX2 based Salae logic clone with 24 MHz max frequency.
I am then unable to check the 50 MHz signal used in the example. By reducing the frequency, the example starts working again.
With lower frequencies (500 kHz, 1,6 MHz, 12 MHz), the SCLK frequency measured with the logic analyzer is correct.

I have also switched the Pico2 board with a RP2040 based Pico board and it works at 50 MHz on current main branch, this is then RP2350 specific.

<details>

<summary>ethernet_w5500_tcp_server.rs version for RP2350</summary>

```rust
//! This example implements a TCP echo server on port 1234 and using DHCP.
//! Send it some data, you should see it echoed back and printed in the console.
//!

#![no_std]
#![no_main]

use defmt::*;
use embassy_executor::Spawner;
use embassy_futures::yield_now;
use embassy_net::{Stack, StackResources};
use embassy_net_wiznet::chip::W5500;
use embassy_net_wiznet::*;
use embassy_rp::block::ImageDef;
use embassy_rp::clocks::RoscRng;
use embassy_rp::gpio::{Input, Level, Output, Pull};
use embassy_rp::peripherals::SPI0;
use embassy_rp::spi::{Async, Config as SpiConfig, Spi};
use embassy_time::{Delay, Duration};
use embedded_hal_bus::spi::ExclusiveDevice;
use embedded_io_async::Write;
use rand::RngCore;
use static_cell::StaticCell;
use {defmt_rtt as _, panic_probe as _};

#[link_section = ".start_block"]
#[used]
pub static IMAGE_DEF: ImageDef = ImageDef::secure_exe();

#[embassy_executor::task]
async fn ethernet_task(
    runner: Runner<
        'static,
        W5500,
        ExclusiveDevice<Spi<'static, SPI0, Async>, Output<'static>, Delay>,
        Input<'static>,
        Output<'static>,
    >,
) -> ! {
    runner.run().await
}

#[embassy_executor::task]
async fn net_task(mut runner: embassy_net::Runner<'static, Device<'static>>) -> ! {
    runner.run().await
}

#[embassy_executor::main]
async fn main(spawner: Spawner) {
    let p = embassy_rp::init(Default::default());

    info!("Start Ethernet");
    let mut rng = RoscRng;
    let mut led = Output::new(p.PIN_25, Level::Low);

    info!("Ethernet init...");
    let mut spi_cfg = SpiConfig::default();
    spi_cfg.frequency = 50_000_000;
    let (miso, mosi, clk) = (p.PIN_20, p.PIN_19, p.PIN_22);
    let spi = Spi::new(p.SPI0, clk, mosi, miso, p.DMA_CH0, p.DMA_CH1, spi_cfg);
    let cs = Output::new(p.PIN_21, Level::High);
    let w5500_int = Input::new(p.PIN_18, Pull::Up);
    let w5500_reset = Output::new(p.PIN_26, Level::High);

    let mac_addr = [0x02, 0x00, 0x00, 0x00, 0x00, 0x00];
    static STATE: StaticCell<State<8, 8>> = StaticCell::new();
    let state = STATE.init(State::<8, 8>::new());
    let (device, runner) = embassy_net_wiznet::new(
        mac_addr,
        state,
        ExclusiveDevice::new(spi, cs, Delay),
        w5500_int,
        w5500_reset,
    )
    .await
    .unwrap();
    unwrap!(spawner.spawn(ethernet_task(runner)));
    info!("Ethernet init done.");

    // Generate random seed
    let seed = rng.next_u64();

    // Init network stack
    static RESOURCES: StaticCell<StackResources<3>> = StaticCell::new();
    let (stack, runner) = embassy_net::new(
        device,
        embassy_net::Config::dhcpv4(Default::default()),
        RESOURCES.init(StackResources::new()),
        seed,
    );

    // Launch network task
    unwrap!(spawner.spawn(net_task(runner)));

    info!("Waiting for DHCP...");
    let cfg = wait_for_config(stack).await;
    let local_addr = cfg.address.address();
    info!("IP address: {:?}", local_addr);

    let mut rx_buffer = [0; 4096];
    let mut tx_buffer = [0; 4096];
    let mut buf = [0; 4096];
    loop {
        let mut socket = embassy_net::tcp::TcpSocket::new(stack, &mut rx_buffer, &mut tx_buffer);
        socket.set_timeout(Some(Duration::from_secs(10)));

        led.set_low();
        info!("Listening on TCP:1234...");
        if let Err(e) = socket.accept(1234).await {
            warn!("accept error: {:?}", e);
            continue;
        }
        info!("Received connection from {:?}", socket.remote_endpoint());
        led.set_high();

        loop {
            let n = match socket.read(&mut buf).await {
                Ok(0) => {
                    warn!("read EOF");
                    break;
                }
                Ok(n) => n,
                Err(e) => {
                    warn!("{:?}", e);
                    break;
                }
            };
            info!("rxd {}", core::str::from_utf8(&buf[..n]).unwrap());

            if let Err(e) = socket.write_all(&buf[..n]).await {
                warn!("write error: {:?}", e);
                break;
            }
        }
    }
}

async fn wait_for_config(stack: Stack<'static>) -> embassy_net::StaticConfigV4 {
    loop {
        if let Some(config) = stack.config_v4() {
            return config.clone();
        }
        yield_now().await;
    }
}

```

</details>

andergisomon, 2025-11-10:
had the same issue, had to bring it down to 10MHz

romainreignier, 2025-11-14:
I have also done some tests on the [Cytron iriv-io](https://www.cytron.io/p-iriv-io-controller) but because the debug pins are not exposed, I have used a Pico2 board on a breadboard with a W5500 breakout board.
On a breadboard, signal integrity might be an issue but when replacing the RP2350 with a RP2040 (Pico board), it works.