embassy-rs/embassy #3531: STM32F411 - problem with serial example

charleselie, 2024-11-12:
Hello,

I'm trying embassy to evaluate differences with the standard use of stm32hal. 

I have difficulties using the serial example.

first : 
use embassy_stm32::usb::{Driver, Instance}; 
usb is not in the root, but usb_otg is, so I've replaced it with usb_otg

second : 
bind_interrupts!(struct Irqs {
    OTG_FS => usb::InterruptHandler<peripherals::USB_OTG_FS>;
});

the interrupt binding is not working and provide the following error : unsafe impl for safe trait and failed to resolve use of undeclared crate or module.

even if I replace with usb_otg, I get the same error.

I've tried different solutions, but can't make the code build, any ideas ?

kingofpayne, 2024-11-14:
Depending on your chip, there might be `USB_OTG_HS` (high speed) instead.

charleselie, 2024-11-14:
I found one part of the problem.
Mainly because I was working with the crate, and not with repo.
After cloning the repo, it "works" better. 

Now my problem is about finding the right configuration to make the USB work, for now I can't pass the configuration, I think it's because of the clock settings. Even if I choose the right Hse, I still can't make it work, but I'll look after that.

if someone has already found a way to make it work with different clock speed than the example.