embassy-rs/embassy #3537: [embassy-sync] Using `Watch`, built with `--release` leads to "Exception 'Load access fault'"

lure23, 2024-11-14:
I started using the new [Watch](https://github.com/embassy-rs/embassy/blob/main/embassy-sync/src/watch.rs) functionality today. Unfortunately, haven't been able to see it work in release builds, yet. This happens:

````
0.813779 [INFO ] Targets powered off and on again.
0.813873 [DEBUG] I2C ADDR: 0 -> 0x29_u7
0.817517 [DEBUG] Ping succeeded: 0xf0,0x02
3.558842 [DEBUG] Board now reachable as: 0x29_u7
3.559000 [DEBUG] Init of board 0 succeeded
3.559229 [INFO ] Init succeeded
3.905247 [ERROR] Exception 'Load access fault' mepc=0x42004a1e, mtval=0x4d421b00
3.905345 [ERROR] TrapFrame { ra: 1107327136, t0: 3905200, t1: 0, t2: 0, t3: 2181074082, t4: 0, t5: 0, t6: 0, a0: 1082556352, a1: 1, a2: 1, a3: 4, a4: 1107315144, a5: 1082171792, a6: 0, a7: 0, s0: 1082577216, s1: 2328, s2: 1082170664, s3: 4, s4: 4096, s5: 1082169368, s6: 0, s7: 1082579628, s8: 1296177920, s9: 1082169344, s10: 1082168999, s11: 1082169304, gp: 1082134720, tp: 0, sp: 1082577136, pc: 1107315230, mstatus: 6273, mcause: 5, mtval: 1296177920 }
3.905661 [ERROR] 0x4200789c
3.905693 [ERROR] 0x42009c40
3.905725 [ERROR] 0x42001264
3.905757 [ERROR] 0x420021b6
3.905789 [ERROR] 0x4200675c
3.905821 [ERROR] 0x42004894
3.905853 [ERROR] 0x42004650
3.905885 [ERROR] 0x4200c312
3.905917 [ERROR] 0x42000132
```

If I change to non-`--release` (wanted to see the symbols in the stack trace), All is Fine.

I presume this is caused by inclusion of `embassy-sync` - I didn't change other dependencies, today.

Don't have a Minimal Repo, but my source is available [here](https://github.com/lure23/Zoo-ESP32_Cx/tree/tof) - note the `tof` branch.

**Steps**

- `git clone git@github.com:lure23/Zoo-ESP32_Cx.git`

   ```
   $ cd Zoo-*
   $ git checkout tof
   ```
- Read the `README` at the top and `tof` levels. 

   This is likely beyond anyone wants to do, to repeat the problem. You'd need to set up tools, bring in vendor C sources... Oh, and you need VL53L5CX hardware! 🫠🎈🎈

- _
    ```
    $ cd tof/vl53l5cx
    $ make -f Makefile.dev m
    ```

**Expected**

The program runs and presents some results.

**Actual**

Above pasted exception.

**Further info**

```
$ make -f Makefile.dev m-debug
[...]
4.313160 [INFO ] Data #0: (30°C, 0ms)
4.313230 [INFO ] .target_status:    [[[Valid, Valid, Valid, SemiValid(9)], [Valid, Valid, SemiValid(9), Valid], [Valid, Valid, Valid, Valid], [Valid, Valid, Valid, Valid]]]
4.313369 [INFO ] .distance_mm:      [[[581, 685, 766, 1002], [850, 781, 997, 1038], [961, 641, 783, 1624], [645, 728, 780, 769]]]
4.404145 [DEBUG] Woke up to INT edge (now low; slept 91ms)
```

That works.

lure23, 2024-11-15:
Updated the bug description on [my own repo](https://github.com/lure23/Zoo-ESP32_Cx/issues/12) (a full rewrite). 

The issue occurs with the combination of:
- `probe-rs`
- `--release` build
- `embassy-sync` - at least `Watch` mechanism, both `.changed()` and `.get()`

peterkrull, 2024-11-26:
Hmm, I have not had any issues with extensive usage (e.g. RP2040, STM32, std, multi-core, interrupts). Could you try replacing the `Watch` with another channel, e.g. a `PubSubChannel` with a depth of 1? Just to see if the crash can be triggered without the `Watch`. Also as far as I can tell `'Load access fault'` has something to do with incorrectly accessed memory, and your `// Something brings in need for global allocator. Fake it!!!` seems a bit spooky to me. Could you try using an actual allocator?

lure23, 2025-02-09:
Closing, since I no longer plan to use `Watch` and "I have not had any issues", i.e. perhaps it was only me...

closed 2025-02-09 by lure23