embassy-rs/embassy #3540: STM32 Sdmmc blocks on write_block

bluespider42, 2024-11-15:
The function `write_block` in the sdmmc module currently blocks while waiting for the SD card after the block has been written. This can be several milliseconds depending of the SD card.
The relevant section of code is:
https://github.com/embassy-rs/embassy/blob/6c4b3d82b637fce5ab6efdc312d7852381d8ddeb/embassy-stm32/src/sdmmc/mod.rs#L1310-L1319
I added a yield and this seems to work well as a quick fix, but I'll admit I'm well out of my depth here to know if there may be undesirable side effects. I guess ideally one would await that entire section and wake on D0 going high - but I don't know if that would be feasible.
```rust
use embassy_futures::yield_now();
...
// Try to read card status (ACMD13)
while timeout > 0 {
    match self.read_sd_status().await {
        Ok(_) => return Ok(()),
        Err(Error::Timeout) => (yield_now().await), // Try again
        Err(e) => return Err(e),
    }
    timeout -= 1;
}
Err(Error::SoftwareTimeout)
```

Attached is the output of a logic analyser (1ms/div) showing the effect of the change (A different type of SD card was used for the two tests which seem to have different wait times).
[image]
[image]

ost-ing, 2025-03-18:
@bluespider42 It seems that `read_sd_status` already polls asynchronously, so in the event of an Error::Timeout it just runs the `read_sd_status` fn again, which would in of itself yield back to the executor, so I'm not really sure this would do all that much. 

Are you using a filesystem? I've personally found that the bottle neck for my projects is that the various fat32 libraries available for Rust are not well optimized.

bluespider42, 2025-03-18:
I'm not using a filesystem, just reading and writing blocks.  
I thought the same, but it does block - see logic analyser output, in the first record I miss several samples on the SPI bus while waiting for `read_sd_status`. I might be showing my async ignorance here but I assumed that the method only actually yields back to the executor if the poll fn returns `Poll::Pending`, and in the case where it immediately returns `Poll::Ready(...)` it just carries on. Ideally the waker would be called on `D0` going high or something but I don't really know how or if that's possible.