embassy-rs/embassy #3560: New release of embassy_stm32?

bifurcation, 2024-11-22:
Is there a plan to update the [version of embassy_stm32](https://crates.io/crates/embassy-stm32) on crates.io?  It hasn't been updated since January, and there have been a bunch of breaking changes.  So the examples no longer build against the release version of the crate, and the documentation on docs.embassy.dev is inaccurate.  The latter point is really annoying, given that docs.embassy.dev is explicitly the recommended source of documentation.

wrs, 2024-12-22:
Not just breaking changes, but important bug fixes like #2833. Is there a planned release cadence to get a new set of mutually-compatible versions into crates.io (and corresponding updated docs) on a regular basis?

Dirbaio, 2025-01-10:
embassy-stm32 v0.2.0 is now released on crates.io

closed 2025-01-10 by Dirbaio