embassy-rs/embassy #3563: stm32wb gatt_server example stops responding after 16 writes to characteristic

crfellows2, 2024-11-22:
First discovered this with a stm32wb55cgu on a custom board, but have noticed it is also an issue with the [gatt_server](https://github.com/embassy-rs/embassy/blob/main/examples/stm32wb/src/bin/gatt_server.rs) example on a stm32 nucleo.

Using two different BLE testing apps (ST BLE Toolbox and my custom app), I can successfully write to the characteristic [`Uuid16(0x502)`](https://github.com/embassy-rs/embassy/blob/main/examples/stm32wb/src/bin/gatt_server.rs#L301) 16 times, but the 17th does not show up in the trace and Bluetooth stops responding entirely.
[trace.txt](https://github.com/user-attachments/files/17876385/trace.txt)

I'm thinking maybe the copro receive buffer is filling up since it works _16_ times and the ipcc receive gets stuck on Pending in [`embassy-stm32/src/ipcc.rs`](https://github.com/embassy-rs/embassy/blob/main/embassy-stm32/src/ipcc.rs)?
```rust
    pub async fn receive<R>(channel: IpccChannel, mut f: impl FnMut() -> Option<R>) -> R {
        let regs = IPCC::regs();

        loop {
            // This is a race, but is nice for debugging
            if !regs.cpu(1).sr().read().chf(channel as usize) {
                trace!("ipcc: ch {}: wait for rx occupied", channel as u8);
            }

            poll_fn(|cx| {
                IPCC::state().rx_waker_for(channel).register(cx.waker());
                // If bit is set to 1 then interrupt is disabled; we want to enable the interrupt
                regs.cpu(0).mr().modify(|w| w.set_chom(channel as usize, false));

                compiler_fence(Ordering::SeqCst);

                if regs.cpu(1).sr().read().chf(channel as usize) {
                    // If bit is set to 1 then interrupt is disabled; we want to disable the interrupt
                    regs.cpu(0).mr().modify(|w| w.set_chfm(channel as usize, true));

                    Poll::Ready(())
                } else {
                    Poll::Pending
                }
            })
            .await;

            trace!("ipcc: ch {}: read data", channel as u8);

            match f() {
                Some(ret) => return ret,
                None => {}
            }

            trace!("ipcc: ch {}: clear rx", channel as u8);
            compiler_fence(Ordering::SeqCst);
            // If the channel is clear and the read function returns none, fetch more data
            regs.cpu(0).scr().write(|w| w.set_chc(channel as usize, true));
        }
    }
```
I'll continue investigating but thought I'd put this out there in case someone more experienced sees an obvious issue.

Sikandar777, 2025-03-04:
If you are using Cubemx, it allows you to change the writing characteristics capability for up to 64

clubby789, 2025-07-23:
https://github.com/embassy-rs/embassy/blob/60ed376a19f8f0401f13f4cdc065452f9cf3ddd7/tests/stm32/src/bin/wpan_mac.rs#L27-L29

This is caused by not running the MemoryManager queue. This is unfortunately omitted from the GATT server example

Sikandar777, 2025-07-23:
can you also share your cubemx ble parameters setting ss?

PR #4447 (2025-07-24): stm32/wb: Add memory manager to GATT example
Closes #3563 - this is necessary to prevent locking up after a certain number of events.