embassy-rs/embassy #3571: Support for ARM7.

Frostie314159, 2024-11-25:
Hi, I wanted to ask if the arm executor supports ARM7, and if not, what steps would be necessary to implement it. 
My use case may seem somewhat ridiculous, but I want to write a firmware for a TI Calypso GSM chipset from I think 1999, and it has an ARM7TDMI core.

Dirbaio, 2024-11-25:
You'll need a `critical-section` implementation. With that, `arch-spin` works on all platforms (but wastes power).

If you want to save power, you can copypaste [this](https://github.com/embassy-rs/embassy/blob/main/embassy-executor/src/arch/riscv32.rs) and change the `wfi` with the equivalent for your chip.

Frostie314159, 2024-11-26:
Thanks! I'll see what I can get running.

closed 2024-11-26 by Frostie314159