embassy-rs/embassy #3590: STM32 ring-buffer

elagil, 2024-11-28:
This is a follow-up to the discussion in https://github.com/embassy-rs/embassy/pull/3336 by @liarokapisv

Feel free to add further people to the discussion.

Remaining issues/ideas with the current ring-buffer implementation include
- `clear()` on overrun causes issues with higher-level drivers that handle data in groups of samples. For example, a `SAI` driver may handle data in groups of two samples (left/right). The ring-buffer has no concept of grouping, so on `clear()` it will cause random read/write index offsets - in the case of the `SAI` example, causing left/right switching.
- Usefulness of `write_immediate()` is not clear, should it be removed?
- The ring buffer may pause automatically on overrun. The ring buffer shall then be explicitly restarted by a user when new data is available.

furrycoding, 2026-09-28:
Maybe I'm using `dma::WriteableRingBuffer` incorrectly(examples are hard to come by), but I find the fact that it's *full on reset* quite weird. For one, it means that effectively non-user-controlled data gets sent at the start, unless `write_immediate` has been called beforehand, passing a buffer of the same size as the ring buffer.

A simple solution seems to be to make another `reset()` method (`reset_empty()`?) that leaves both read and write index at zero - this would allow the buffer to be hydrated using the same `write()` or `write_exact()` methods before starting the DMA and ensure that no old buffer data is unexpectedly sent out

elagil, 2026-09-28:
You may also have a look at this PR, where this is discussed as well: https://github.com/embassy-rs/embassy/pull/7070.

Yes, `write_immediate()` is just a workaround for the issue that you mention. Making the ringbuffer start empty seems to be better.

wojciech-graj, 2026-10-07:
From #7070:
> I support the ringbuffer starting empty and failing with overrun on start() immediately.

A DMA request may be issued at any time, and probably not *immediately* after start() is called. As such, if start() could be called on an empty ringbuffer without erroring out, one could start the DMA, stream data to it at a later point, and this would be fine as long as a DMA request hadn't been issued by then.

Many systems will not have any data to fill the ringbuffer at construction. Instead of calling start() at construction, the user would have to check if the ringbuffer is already started and start it if not after every write.

If this is implemented, I'd suggest having it be togglable.

elagil, 2026-10-09:
Just so you know, a PR that creates the ringbuffer empty was already merged: https://github.com/embassy-rs/embassy/pull/7159

Not immediately failing after `start()` if no DMA request comes in seems fair, but how do you catch an underrun then? On the next `write()`, or should it panic then?

wojciech-graj, 2026-10-10:
We currently return an error on the next write in case of an underrun, and it would be quite odd to have different behavior exclusively for the case where the ringbuffer was empty. But I also generally believe panics should only happen when an error is unrecoverable, which this is not.

elagil, 2026-10-10:
I am a bit biased, because I use DMA a lot with I2S/SAI and it will certainly fail immediately after `start()` because the peripheral will start clocking right away. But failing on next `write()` is good too.