embassy-rs/embassy #3596: Create `DynamicSender` from `DynamicReceiver` and vice versa.

Frostie314159, 2024-12-01:
Hi, for a rather specific use case, it would be nice to acquire a `DynamicSender` from a `DynamicReceiver`, since this would allow eliding a lot of generics. Is there anything preventing this, or a design choice speaking against this? If not, I would be happy to make a PR implementing that.

Dirbaio, 2024-12-01:
Having this feature would "feel wrong". It's nice to be able to rely on the fact that if you give some code a Receiver it can only receive, it can't magically convert that to a Sender and start sending into the channel.

Frostie314159, 2024-12-01:
Okay, that's my thought process too. Another option would be, to make a `DynChannel`, which exposes both methods. Maybe that's more idiomatic.