embassy-rs/embassy #3604: embassy-net-nrf91 panics during tcp retransmission

mdrssv, 2024-12-03:
Hi,

I've been toying around with the nrf9160 example using a [Thingy91](https://www.nordicsemi.com/Products/Development-hardware/Nordic-Thingy-91).
However when running the program for some period of time I almost always run into an panic due to `NoFreeBufs`. My guess would be, that the retransmission generates to many packets(one packet = one `TX_BUF`) which cannot be written out fast enough, thus exhausting the 4 defined buffers.

Modified main, otherwise everthing is pretty much like in the [original example](https://github.com/embassy-rs/embassy/blob/main/examples/nrf9160/src/bin/modem_tcp_client.rs)
```rust
#[embassy_executor::main]
async fn main(spawner: Spawner) {
    let p = embassy_nrf::init(Default::default());

    info!("Hello World!");

    let ipc_mem = unsafe {
        let ipc_start = &__start_ipc as *const u8 as *mut MaybeUninit<u8>;
        let ipc_end = &__end_ipc as *const u8 as *mut MaybeUninit<u8>;
        let ipc_len = ipc_end.offset_from(ipc_start) as usize;
        slice::from_raw_parts_mut(ipc_start, ipc_len)
    };

    static mut TRACE_BUF: [u8; 4096] = [0u8; 4096];
    let mut config = uarte::Config::default();
    config.baudrate = Baudrate::BAUD1M;
    let uart = BufferedUarteTx::new(
        //let trace_uart = BufferedUarteTx::new(
        unsafe { peripherals::SERIAL0::steal() },
        Irqs,
        unsafe { peripherals::P0_01::steal() },
        //unsafe { peripherals::P0_14::steal() },
        config,
        unsafe { &mut *addr_of_mut!(TRACE_BUF) },
    );

    static STATE: StaticCell<State> = StaticCell::new();
    static TRACE: StaticCell<TraceBuffer> = StaticCell::new();
    let (device, control, runner, tracer) = embassy_net_nrf91::new_with_trace(
        STATE.init(State::new()),
        ipc_mem,
        TRACE.init(TraceBuffer::new()),
    )
    .await;
    unwrap!(spawner.spawn(modem_task(runner)));
    unwrap!(spawner.spawn(trace_task(uart, tracer)));

    let led = |_, _, _| {};
    #[cfg(feature = "led")]
    let mut led = {
        let mut led_pwm = SimplePwm::new_3ch(p.PWM0, p.P0_29, p.P0_30, p.P0_31);
        led_pwm.set_period(16_000_000);
        led_pwm.set_max_duty(255);
        for ch in 0..3 {
            led_pwm.set_duty(ch, 255);
        }
        led_pwm.enable();
        let mut led = move |r: u8, g: u8, b: u8| {
            let max_duty = led_pwm.max_duty() as u32;
            let duty = |val: u8| (val as u32 * max_duty / 255u32) as u16;
            for (ch, dc) in [r, g, b].into_iter().enumerate() {
                led_pwm.set_duty(ch, duty(dc));
            }
        };
        led(10, 10, 10);
        led
    };

    let config = embassy_net::Config::default();

    // Generate "random" seed. nRF91 has no RNG, TODO figure out something...
    let seed = 1;

[... 505 lines cut ...]
INFO  tx buf 3
└─ embassy_net_nrf91::{impl#3}::send_message @ /home/marvin/.cargo/git/checkouts/embassy-9312dcb0ed774b29/04c1d69/embassy-net-nrf91/src/lib.rs:515
TRACE using tx msg idx 0
└─ embassy_net_nrf91::{impl#3}::find_free_message @ /home/marvin/.cargo/git/checkouts/embassy-9312dcb0ed774b29/04c1d69/embassy-net-nrf91/src/lib.rs:489
DEBUG tx seq 214 msg: Message { id: 1879441412, channel: 2, unk1: 0, unk2: 0, unk3: 0, data: 0x20001afc, data_len: 55, param_len: 12, param: [0, 0, 0, 0, 128, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0] }
└─ embassy_net_nrf91::{impl#3}::send_message_raw @ /home/marvin/.cargo/git/checkouts/embassy-9312dcb0ed774b29/04c1d69/embassy-net-nrf91/src/lib.rs:539
TRACE irq
└─ embassy_net_nrf91::on_ipc_irq @ /home/marvin/.cargo/git/checkouts/embassy-9312dcb0ed774b29/04c1d69/embassy-net-nrf91/src/lib.rs:41
TRACE poll!
└─ embassy_net_nrf91::{impl#3}::poll @ /home/marvin/.cargo/git/checkouts/embassy-9312dcb0ed774b29/04c1d69/embassy-net-nrf91/src/lib.rs:321
TRACE ipc 4
└─ embassy_net_nrf91::{impl#3}::poll @ /home/marvin/.cargo/git/checkouts/embassy-9312dcb0ed774b29/04c1d69/embassy-net-nrf91/src/lib.rs:355
DEBUG rx seq 319 msg: Message { id: 2147876868, channel: 2, unk1: 0, unk2: 0, unk3: 0, data: 0x0, data_len: 0, param_len: 16, param: [0, 0, 0, 0, 128, 0, 0, 0, 0, 0, 0, 0, 55, 0, 0, 0, 2, 0, 0, 0, 200, 34, 0, 32, 16, 0, 0, 0, 4, 0, 0, 0, 184, 0, 0, 0, 4, 0, 1, 144, 2, 0, 0, 0] }
└─ embassy_net_nrf91::{impl#3}::process @ /home/marvin/.cargo/git/checkouts/embassy-9312dcb0ed774b29/04c1d69/embassy-net-nrf91/src/lib.rs:468
WARN  no free tx buf
└─ embassy_net_nrf91::{impl#3}::find_free_tx_buf @ /home/marvin/.cargo/git/checkouts/embassy-9312dcb0ed774b29/04c1d69/embassy-net-nrf91/src/lib.rs:503
ERROR panicked at 'unwrap failed: state.send_message(msg, req_data)'
error: `NoFreeBufs`
└─ embassy_net_nrf91::{impl#5}::request::{async_fn#0} @ /home/marvin/.cargo/git/checkouts/embassy-9312dcb0ed774b29/04c1d69/embassy-net-nrf91/src/lib.rs:762
ERROR panicked at /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/defmt-0.3.10/src/lib.rs:380:5:
explicit panic
└─ panic_probe::print_defmt::print @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/panic-probe-0.3.2/src/lib.rs:104

```

lulf, 2024-12-03:
Hi, could you try the fixes in this PR? https://github.com/embassy-rs/embassy/pull/3603

mdrssv, 2024-12-03:
> Hi, could you try the fixes in this PR? #3603

Still happening
```
DEBUG retransmitting at t+0.796s
└─ smoltcp::socket::tcp::{impl#9}::dispatch @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:18
TRACE rtte: abort sampling due to retransmit
└─ smoltcp::socket::tcp::{impl#6}::on_retransmit @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE outgoing segment will send data or flags
└─ smoltcp::socket::tcp::{impl#9}::dispatch @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE tx buffer: sending 5 octets at offset 0
└─ smoltcp::socket::tcp::{impl#9}::dispatch @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE sending PSH|ACK
└─ smoltcp::socket::tcp::{impl#9}::dispatch @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE tx buffer: dequeueing 5 octets (now 0)
└─ smoltcp::socket::tcp::{impl#9}::process @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE rx buffer: receiving 4 octets at offset 0
└─ smoltcp::socket::tcp::{impl#9}::process @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE rx buffer: enqueueing 4 octets (now 24)
└─ smoltcp::socket::tcp::{impl#9}::process @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE starting delayed ack timer
└─ smoltcp::socket::tcp::{impl#9}::process @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE outgoing segment will send data or flags
└─ smoltcp::socket::tcp::{impl#9}::dispatch @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE tx buffer: sending 5 octets at offset 0
└─ smoltcp::socket::tcp::{impl#9}::dispatch @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE sending PSH|ACK
└─ smoltcp::socket::tcp::{impl#9}::dispatch @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE stop delayed ack timer
└─ smoltcp::socket::tcp::{impl#9}::dispatch @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE rtte: sampling at seq=22509129
└─ smoltcp::socket::tcp::{impl#6}::on_send @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE rtte: sample=49 rtt=224 dev=147 rto=812
└─ smoltcp::socket::tcp::{impl#6}::sample @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE tx buffer: dequeueing 5 octets (now 0)
└─ smoltcp::socket::tcp::{impl#9}::process @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE rx buffer: receiving 1 octets at offset 0
└─ smoltcp::socket::tcp::{impl#9}::process @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE rx buffer: enqueueing 1 octets (now 20)
└─ smoltcp::socket::tcp::{impl#9}::process @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE starting delayed ack timer
└─ smoltcp::socket::tcp::{impl#9}::process @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE outgoing segment will acknowledge
└─ smoltcp::socket::tcp::{impl#9}::dispatch @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE sending ACK
└─ smoltcp::socket::tcp::{impl#9}::dispatch @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE stop delayed ack timer
└─ smoltcp::socket::tcp::{impl#9}::dispatch @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE rx buffer: receiving 5 octets at offset 0
└─ smoltcp::socket::tcp::{impl#9}::process @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE rx buffer: enqueueing 5 octets (now 20)
└─ smoltcp::socket::tcp::{impl#9}::process @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE starting delayed ack timer
└─ smoltcp::socket::tcp::{impl#9}::process @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE outgoing segment will acknowledge
└─ smoltcp::socket::tcp::{impl#9}::dispatch @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE sending ACK
└─ smoltcp::socket::tcp::{impl#9}::dispatch @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
TRACE stop delayed ack timer
└─ smoltcp::socket::tcp::{impl#9}::dispatch @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:17
ERROR panicked at 'unwrap failed: state.send_message(msg, req_data)'
error: `NoFreeBufs`
└─ embassy_net_nrf91::{impl#5}::request::{async_fn#0} @ /home/marvin/.cargo/git/checkouts/embassy-9312dcb0ed774b29/6065bc6/embassy-net-nrf91/src/lib.rs:765
ERROR panicked at /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/defmt-0.3.10/src/lib.rs:380:5:
explicit panic
└─ panic_probe::print_defmt::print @ /home/marvin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/panic-probe-0.3.2/src/lib.rs:104
```

lulf, 2024-12-04:
@mdrssv I've not gotten a public tcp server setup yet to reproduce the retransmits, and the reproducer have been running for a while. However, I've got another patch here https://github.com/embassy-rs/embassy/pull/3609 - It should make it wait for free bufs instead of failing.

mdrssv, 2024-12-05:
> @mdrssv I've not gotten a public tcp server setup yet to reproduce the retransmits, and the reproducer have been running for a while. However, I've got another patch here #3609 - It should make it wait for free bufs instead of failing.

Did zou try an loop like this one?

```
for _ in 0..3 {
     socket.write_all(b"ping\n").await?;
     socket.read(&mut buf[..5]).await?;
}
```

If I reduce the numer of iterations to 1 the whole thing will run a lot longer whereas 3 iterations cause an panic on the first run

lulf, 2024-12-06:
> Did zou try an loop like this one?
>
> ```
> for _ in 0..3 {
>      socket.write_all(b"ping\n").await?;
>      socket.read(&mut buf[..5]).await?;
> }
> ```
>
> If I reduce the numer of iterations to 1 the whole [...]

Yeah, I've copied the code as is (nit: the read() should ideally be read_exact() to ensure 5 bytes are read). I've also increased to use 10 iterations, and using 1 sec as the interval. I've setup a TCP server on a public host and using traffic control on linux to introduce packet loss. So far I've been able to trigger the NoFreeBufs in the Runner, so it should be possible to trigger it in the request() path as well I hope. 

I'm trying now to enable some code that polls control.status() every sec to improve chances that the unwrap() in request() is triggered. If I can get that to trigger, then I will try with the changes in #3609 as well.

lulf, 2024-12-06:
I'm still not able to reproduce this even with dropping most packets on server side, checking control.status() often to trigger requests etc. It would be great if you could test with #3609 as well.

closed 2025-07-08 by Dirbaio