embassy-rs/embassy #3611: How to set task-arena-size for ESP

tylantz, 2024-12-04:
I am running into an issue in which I seemingly cannot set a large enough task arena for a very simple app.  Because I am writing code for the ESP32s3, I have to use channel= "esp" and presumably cannot use "nightly" to benefit from compiler-set arena sizing. 

My code is very simple, it establishes a wifi connection using the example code is [esp-hal](https://github.com/esp-rs/esp-hal/blob/main/examples/src/bin/wifi_embassy_dhcp.rs) and then makes a web request using [reqwless](https://github.com/drogue-iot/reqwless) in the main function. When the function call to make the web request is included and called in my main function, I get the error: ```embassy-executor: task arena is full. You must increase the arena size, see the documentation for details: https://docs.embassy.dev/embassy-executor/```. After setting the arena size as high as 393216 using the cargo.toml feature flag, I am still getting panics.

The docs do not provide guidance on how to set the arena size. What does the arena size relate to? How does one determine what is an appropriate size? Any idea why I am getting these panics at what I assume is a large arena size for a simple application?

Thanks for the help!

Dirbaio, 2024-12-05:
> I have to use channel= "esp" and presumably cannot use "nightly" to benefit from compiler-set arena sizing.

the espressif xtenssa rust fork does have nightly features enabled, so you should be able to enable the `nightly` Cargo feature on embassy-executor. I recommend doing that, so you don't hae to tune the arena size at all.

> The docs do not provide guidance on how to set the arena size. What does the arena size relate to?

it's documented here https://docs.embassy.dev/embassy-executor/git/cortex-m/index.html#task-arena

> How does one determine what is an appropriate size? Any idea why I am getting these panics at what I assume is a large arena size for a simple application?

it has to be larger than the sum of sizes for the executor. It's a bit hard to know that, so in practice the easiest is if you got panics then just bump it up.

393kb is *a lot* though. Maybe it's not taking effect for some reason? If you're using `embassy-executor` through some `esp32-*` crate, make sure you're using the exact same version. Otherwise you're setting the size on one embassy-executor version and `esp32-*` keeps using the defaults because it uses another version.

tylantz, 2024-12-05:
Hi Daario, 

Thanks for the quick reply. I appreciate the help!

> the espressif xtenssa rust fork does have nightly features enabled, so you should be able to enable the nightly Cargo feature on embassy-executor. I recommend doing that, so you don't have to tune the arena size at all.  

The program runs perfectly after changing the feature flag from a ```task-arena-size-[size]``` to ```nightly```. 

> it's documented here https://docs.embassy.dev/embassy-executor/git/cortex-m/index.html#task-arena

You are absolutely right. I wasn't clear. I meant to say: setting the value via feature flag or environment variable is well-documented, but determining the appropriate value is not. It sounds like it is more art than science so trial-and-error or using nightly features is the way to go.   

> Maybe it's not taking effect for some reason? If you're using embassy-executor through some esp32-* crate, make sure you're using the exact same version. Otherwise you're setting the size on one embassy-executor version and esp32-* keeps using the defaults because it uses another version.

I tried setting the value both by feature flag and by environment variable. I don't know the internals of the executor to check if the flag is set, however, I did try setting two different arena sizes via feature flags and I got the expected compile error. I checked cargo tree and the two instances of the executor are using the same version. 

```
├── embassy-executor v0.6.3 task-arena-size-393216
│   ├── embassy-executor-macros v0.6.2 (proc-macro) 
│   ...
├── esp-hal-embassy v0.4.0 default,esp32s3,executors
│   ├── embassy-executor v0.6.3 task-arena-size-393216 (*)
|   ...
```

My problem is fixed with the nightly flag, but let me know if you want me to check anything else or share the code. I am happy to help trouble shoot this for anyone else who may run into it in the future.

bugadani, 2024-12-08:
FWIW the way to specify the arena size should just work as usual, but I don't know if the S3 has a contiguous memory area enough for the 393k. You didn't specify what panics you got after setting it, so it looks like it's still the "arena is full" error, but it sounds somewhat implausible.

closed 2025-07-08 by Dirbaio