embassy-rs/embassy #3615: [embassy-rp] pwm::Pwm drop leaves gpios in incorrect state

Jojo-1000, 2024-12-05:
Hi, I am using a Pwm to control some motors on Pico 2.

I noticed that when the Pwm goes out of scope, it leaves them running. I looked at the drop implementation and there seems to be a difference between `Pwm` and `PwmOutput`. Using the two channels split as `PwmOutput` works as expected. Here is the implementation for reference:

This one works as expected:
https://github.com/embassy-rs/embassy/blob/af02310edc60c568368d2e8805bb07a0797723a8/embassy-rp/src/pwm.rs#L398-L427

This one leaves the pins in an incorrect state, the part about pull down is missing:
https://github.com/embassy-rs/embassy/blob/af02310edc60c568368d2e8805bb07a0797723a8/embassy-rp/src/pwm.rs#L485-L495

## Reproduction
Unexpected:
```rs
{
    let mut c = embassy_rp::pwm::Config::default();
    let pwm_freq = 20_000;
    let clock_freq = embassy_rp::clocks::clk_sys_freq();
    c.top = (clock_freq / pwm_freq) as u16 - 1;
    
    let mut pwm = embassy_rp::pwm::Pwm::new_output_ab(pwm_slice6, pin_11, pin_10, c.clone());
    pwm.set_duty_cycle_fully_off().unwrap();
}
// There is an output signal
```
Expected:
```rs
{
    let mut c = embassy_rp::pwm::Config::default();
    let pwm_freq = 20_000;
    let clock_freq = embassy_rp::clocks::clk_sys_freq();
    c.top = (clock_freq / pwm_freq) as u16 - 1;
    
    let pwm = embassy_rp::pwm::Pwm::new_output_ab(pwm_slice6, pin_11, pin_10, c.clone());
    let (pwm_a,pwm_b) = pwm.split();
    pwm_a.unwrap().set_duty_cycle_fully_off().unwrap();
    pwm_b.unwrap().set_duty_cycle_fully_off().unwrap();
}
// There is no output signal
```

1-rafael-1, 2025-04-13:
@vivi202 since this was probably my fault i today tried to reproduce this - without success. I used gpio25 for convenience, but no way could i see the led still lighting up after leaving scope or calling drop. Can You still repro?

vivi202, 2025-04-13:
hi i'm not the autor of this issue i implemented `pwm.split()`. 
i wrote a fix for this issue but i did not have the time to  test.

1-rafael-1, 2025-04-14:
> hi i'm not the autor of this issue, i implemented `pwm.split()`. i wrote a fix for this issue but i did not have the time to test.

ah okay! :)

@Jojo-1000 I tried to reproduce with your snippets, but did not manage to observe the behavior. I had to alter from slice6 to slice5, because 10 and 11 are on slice5 on Pico 2. I put a scope on pin 11 and when i drop the pwm, the signal stops.  and the pin goes low. Am I missing something?

Jojo-1000, 2025-04-14:
@1-rafael-1 It has been a while since I had this problem, so I don't remember exactly which code I used to reproduce this problem. I originally had this problem in a bigger code base with some other stuff happening in parallel, but I am pretty sure I tried the reduced snippet here and it had the same trouble. I used PWM_SLICE6 with pin 12/13 and PWM_SLICE5 with pin 10/11 if you want to try the other one, it seems that got mixed up in the snippet. Right now I don't have access to the Pico, but I will check it in a week or so and see if I can reproduce it then.

My circuit basically uses the PWM output to drive a transistor motor driver, and during prototyping the motor would keep spinning even though the PWM went out of scope. As soon as I split the output it stopped happening. I don't have a scope or debugger to give too much more insight on this. I just looked at the code and saw this difference which explained it for me.

1-rafael-1, 2025-04-14:
@Jojo-1000 I believe there have been other changes to the HAL since after this issue was posted, maybe it got fixed by some other change. Anyway I will have another go at trying a repro some other day, maybe also using a rp2040 for comparison. Then I am at my current skill levels wits end, if i cannot repro :)

vivi202, 2025-04-14:
Can you try my fix?, I think it's better to enable pin pulldown and clear the registers after the slice is disabled to restore the pin to default state. 
I remember that when I have implemented `pwm.split()` if i didn't enable the pulldown. I had some issue. If it works we can propose to merge that as an improvement.

vivi202, 2025-04-14:
> @Jojo-1000 I believe there have been other changes to the HAL since after this issue was posted, maybe it got fixed by some other change. Anyway I will have another go at trying a repro some other [...]

Maybe it's noise coupling between the pins because they are floating he was using two slices.

Jojo-1000, 2025-04-15:
@1-rafael-1 I think for reproduction of this issue it would be sufficient to compare the relevant pin registers to the default state according to the datasheet. It would be my expectation that it is restored to default, which includes pull down if I remember correctly.

1-rafael-1, 2025-04-15:
@Jojo-1000 @vivi202 Agreed. Had a look briefly today and I think I know why I could not repro, the scope and also the onboard led will not be (visibly) affected from floating pin I think but the motor driver might have other much lower thresholds. So, given I will not observe a difference and also given the PwmOutput already has the PDE on drop and given that the proposed change has already been proven to work and lastly given that  the reasoning behind it sounds convincing: There is nothing I can add at my current skill level and I would suggest to propose the change. @vivi202 I do not think a test from me will add anything, I have no motor driver and motor on hand and the scope showed nothing without the fix and will show nothing with it.