embassy-rs/embassy #3674: STM32L5 button_exti example panic

DBY047, 2024-12-21:
I loaded the button exti example in my NUCLEO-L552ZE-Q dev-kit and get a panic error when I push the button:
```bash
2.353698 ERROR panicked at /home/danny/dev-test/embassy/embassy-stm32/src/low_power.rs:97:23:
called `Option::unwrap()` on a `None` value
```
I changed the code on_wakup_irq function to match the as_mut() result to fix the issue.
```rust
#[allow(dead_code)]
pub(crate) unsafe fn on_wakeup_irq() {
    match EXECUTOR.as_mut() {
        Some(executor) => executor.on_wakeup_irq(),
        None => {}
    }
}
```

Dirbaio, 2024-12-21:
try disabling the `low-power` feature for `embassy-stm32` in Cargo.toml.

DBY047, 2024-12-21:
It worked. Thanks.

closed 2024-12-21 by DBY047