embassy-rs/embassy #3677: No rp2040, rp235xa, or rp235xb features

joshuafocht, 2024-12-22:
When trying to switch between rp2040 and rp235xa/b there is no feature to do so.
Am I trying to do something wrong? Here is the dependencies section of my cargo.toml
```toml
[dependencies]
defmt = "0.3"
defmt-rtt = "0.4"
panic-probe = { version = "0.3", features = ["print-defmt"] }

embedded-hal = "1.0.0"
embedded-hal-async = "1.0.0"
embedded-io = "0.6.1"
embedded-io-async = "0.6.1"
embedded-storage = "0.3.1"

cortex-m-rt = "0.7.5"

embassy-executor = { version = "0.6", features = ["task-arena-size-1024", "arch-cortex-m", "executor-thread", "defmt", "integrated-timers", "executor-interrupt"] }
embassy-sync = { version = "0.6" }
embassy-time = { version = "0.3", features = ["defmt", "defmt-timestamp-uptime"] }

cortex-m = { version = "0.7.7" }
embassy-rp = { version = "0.2", features = ["defmt", "unstable-pac", "time-driver", "critical-section-impl", "rp235xa"] }
```

CBJamo, 2024-12-27:
The embassy-rp version released to crates is somewhat old. Switch to git by adding a[ patch section to your cargo.toml](https://embassy.dev/book/index.html#_how_do_i_switch_to_the_main_branch).

Dirbaio, 2025-01-10:
embassy-rp v0.3.0 is now released on crates.io

closed 2025-01-10 by Dirbaio