embassy-rs/embassy #3681: I2S write future never completes in `examples/stm32f4/.../i2s_dma.rs`

vinsynth, 2024-12-23:
I'm using an stm32f411ceu6 chip. Having replaced the necessary chip definitions in `.cargo/config.toml` and `Cargo.toml` (such that the `blinky.rs` program compiles and runs as expected), and remapped pins & DMA streams/channels as necessary, I find that `i2s.write(...)` in `i2s_dma.rs` never completes. Here's my slightly modified code in case anyone can try this on their own chip:
```
#![no_std]
#![no_main]

use defmt::*;
use embassy_executor::Spawner;
use embassy_stm32::i2s::{Config, I2S};
use embassy_stm32::time::Hertz;
use {defmt_rtt as _, panic_probe as _};

#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    let p = embassy_stm32::init(Default::default());
    info!("Hello World!");

    let mut dma_buffer = [0x00_u16; 128];
    let mut i2s = I2S::new_txonly(
        p.SPI3,
        p.PB5,  // sd
        p.PA15, // ws
        p.PB3,  // ck
        p.PB10, // mck
        p.DMA1_CH7,
        &mut dma_buffer,
        Hertz(48_000),
        Config::default(),
    );

    for i in 0_u16.. {
        i2s.write(&mut [i * 2; 64]).await.ok();
        defmt::panic!("if this message is reached, the write function returned!");
        i2s.write(&mut [i * 2 + 1; 64]).await.ok();
    }
}
```
`i2s.write(...)` still doesn't complete when `I2S::new_txonly_nomck(...)` is used instead. I'm connecting the pins to a PCM5102A chip that has worked with other i2s drivers. Is there something I'm missing with my configuration, or is this a library issue?

elagil, 2025-01-02:
`i2s.start()` is never called, and I think it should.

elagil, 2025-01-02:
Another thing I noticed is that I2S calculates its clock based on APB1 clock (default 42 MHz) which usually doesn't match the PLLI2S clock. This needs to be fixed at some point.

vinsynth, 2025-01-02:
I tried calling `i2s.start()` before the write statement; the write future still doesn't complete. Calling it after the write statement also doesn't work (it would never make it there anyhow). I also thought of storing the write future to a temporary variable, then calling `i2s.start()`, then awaiting the temporary future, but that uses multiple mutable borrows at once, which rust of course doesn't allow. Unless there's another place to call `i2s.start()` that I'm missing, that function has no effect here.

elagil, 2025-01-02:
By default, `plli2s` is `None`, so you have no clock for the peripheral.

Try something like

```rust
    let mut peripheral_config = embassy_stm32::Config::default();
    {
        // Uses a 25 MHz external oscillator.
        use embassy_stm32::rcc::*;
        peripheral_config.rcc.hse = Some(Hse {
            freq: Hertz(25_000_000),
            mode: HseMode::Oscillator,
        });
        peripheral_config.rcc.sys = Sysclk::PLL1_P;

        peripheral_config.rcc.ahb_pre = AHBPrescaler::DIV1;
        peripheral_config.rcc.apb1_pre = APBPrescaler::DIV2;
        peripheral_config.rcc.apb2_pre = APBPrescaler::DIV1;

        peripheral_config.rcc.mux.clk48sel = mux::Clk48sel::PLL1_Q;

        peripheral_config.rcc.pll_src = PllSource::HSE;
        peripheral_config.rcc.pll = Some(Pll {
            prediv: PllPreDiv::DIV25,
            mul: PllMul::MUL336,
            divp: Some(PllPDiv::DIV4),
            divq: Some(PllQDiv::DIV7),
            divr: None,
        });

        peripheral_config.rcc.plli2s = Some(Pll {
            prediv: PllPreDiv::DIV25,
            mul: PllMul::MUL384,
            divp: None,
            divq: None,
            divr: Some(PllRDiv::DIV5),
        });

        peripheral_config.enable_debug_during_sleep = true;
    }
    let p = embassy_stm32::init(peripheral_config);
```

vinsynth, 2025-01-04:
Aha, that did it! The default clock configuration must've changed after the example was created. I can create a PR to update the example.

elagil, 2025-01-04:
Great! I could also add it here: https://github.com/embassy-rs/embassy/pull/3716

It fixes the broken clock setup on F4 (but not other series chips, they are also broken).

vinsynth, 2025-01-20:
Nice! I've noticed another oddity, however.

Referencing the recommended I2S clock configurations in Table 90 of the STM32F411CEU6 reference manual, I extended the example file to what's shown below, with the goal of configuring the I2S bus for a sample rate of 48 kHz:
```
#![no_std]
#![no_main]

use defmt::*;
use embassy_executor::Spawner;
use embassy_stm32::i2s::{Config, I2S};
use embassy_stm32::time::Hertz;
use {defmt_rtt as _, panic_probe as _};

#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    let config = {
        use embassy_stm32::rcc::*;

        let mut config = embassy_stm32::Config::default();
        config.rcc.hse = Some(Hse {
            freq: Hertz::mhz(25),
            mode: HseMode::Oscillator,
        });
        config.rcc.pll_src = PllSource::HSE;
        config.rcc.pll = Some(Pll {
            prediv: PllPreDiv::DIV25,
            mul: PllMul::MUL192,
            divp: Some(PllPDiv::DIV2),
            divq: Some(PllQDiv::DIV4),
            divr: None,
        });
        config.rcc.sys = Sysclk::PLL1_P;
        config.rcc.mux.clk48sel = mux::Clk48sel::PLL1_Q;

        config.rcc.ahb_pre = AHBPrescaler::DIV1;
        config.rcc.apb1_pre = APBPrescaler::DIV2;
        config.rcc.apb2_pre = APBPrescaler::DIV1;

        config.rcc.plli2s = Some(Pll {
            prediv: PllPreDiv::DIV25,
            mul: PllMul::MUL258,
            divp: None,
            divq: None,
            divr: Some(PllRDiv::DIV3),
        });
        config.enable_debug_during_sleep = true;

        config
    };

    let p = embassy_stm32::init(config);

    let _mco = embassy_stm32::rcc::Mco::new(
        p.MCO1,
        p.PA8,
        embassy_stm32::rcc::Mco1Source::HSE,
        embassy_stm32::rcc::McoPrescaler::DIV1,
    );

    let mut dma_buffer = [0x00_u16; 128];
    let mut i2s = I2S::new_txonly(
        p.SPI3,
        p.PB5,  // sd
        p.PA15, // ws
        p.PB3,  // ck
        p.PB10, // mck
        p.DMA1_CH7,
        &mut dma_buffer,
        Hertz(48_000),
        Config::default(),
    );
    i2s.start();

    for i in 0_u16.. {
        i2s.write(&[i * 2; 64]).await.ok();
        i2s.write(&[i * 2 + 1; 64]).await.ok();
    }
}
```
Using some print debugging, I verified that the clock configuration in `I2S::new_inner(...)` does correctly calculate an `I2SDIV` of 3 and `I2SODD` of 1; despite this, an external oscilloscope instead measures a frequency of around 37.5 kHz based off the WS clock (which is audibly present when outputting a basic square wave). Using MCO1, I've confirmed that the HSE used for PLLI2S is operating at 25 MHz, as expected. No external I2S-accepting chip is connected to the board, and the effect persists when using `I2S::new_txonly_nomck(...)`. Am I still missing something?

elagil, 2025-01-20:
Hm, I will try and replicate that case.

elagil, 2025-02-02:
Can you reach perfect clocks for 48 kHz with those I2S PLL settings? I think you might not.

You could try

```
mul: PllMul::MUL384
divr: Some(PllRDiv::DIV5
```

with otherwise equal setup.

vinsynth, 2025-02-02:
The manual recommends the config I used for 48 kHz when generating a master clock—the config you suggest is recommended when a master clock isn’t being generated. Using `I2S::new_txonly_nomck(…)` with the config you suggest produced the same slower output as my setup in my original testing.

elagil, 2025-02-02:
Sorry, I cannot reproduce with my clock settings as above. I use STM32F401, though.

Here are my word clock and bit clock (32 bit per sample).
I run a 24.576 MHz oscillator, but used your PLL settings for demonstration - therefore clocks are too slow by a factor of 1.01725...

Expected clocks are 47.18592 kHz word clock and 3.0199 MHz bit clock.

[image]
[image]

elagil, 2025-02-02:
Here is the full code: https://github.com/blus-audio/firmware-rs/blob/b32e98a649f27da85fb61965af16bbcab00228a7/blus_mini_mk1/src/main.rs#L112

vinsynth, 2025-02-02:
I notice your full code configures other clock-dependent peripherals (e.g., USB) in addition to i2s. Are you able to try running the simpler code I presented earlier (with pin remappings as is appropriate) on your device? If that works as expected, I can try my code on another stm32f411 chip and see if it presents the same issue. My suspicion then would be that something is different with how I2S is configured for f401 vs f411 chips.

vinsynth, 2025-02-02:
Actually, I just now tried running my code on another f411 chip--same slowdown by the same factor as that I experienced with the first f411 chip.

elagil, 2025-02-02:
I used your code with minor changes:
- Had to lower sysclock
- Change pins

My clocks are correct.

```rust
#![no_std]
#![no_main]

use embassy_executor::Spawner;
use embassy_stm32::i2s::{Config, I2S};
use embassy_stm32::time::Hertz;
use {defmt_rtt as _, panic_probe as _};

#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    let config = {
        use embassy_stm32::rcc::*;

        let mut config = embassy_stm32::Config::default();
        config.rcc.hse = Some(Hse {
            freq: Hertz::mhz(25),
            mode: HseMode::Bypass,
        });
        config.rcc.pll_src = PllSource::HSE;
        config.rcc.pll = Some(Pll {
            prediv: PllPreDiv::DIV25,
            mul: PllMul::MUL192,
            divp: Some(PllPDiv::DIV4),
            divq: Some(PllQDiv::DIV8),
            divr: None,
        });
        config.rcc.sys = Sysclk::PLL1_P;
        config.rcc.mux.clk48sel = mux::Clk48sel::PLL1_Q;

        config.rcc.ahb_pre = AHBPrescaler::DIV1;
        config.rcc.apb1_pre = APBPrescaler::DIV2;
        config.rcc.apb2_pre = APBPrescaler::DIV1;

        config.rcc.plli2s = Some(Pll {
            prediv: PllPreDiv::DIV25,
            mul: PllMul::MUL258,
            divp: None,
            divq: None,
            divr: Some(PllRDiv::DIV3),
        });
        config.enable_debug_during_sleep = true;

        config
    };

    let p = embassy_stm32::init(config);

    let _mco = embassy_stm32::rcc::Mco::new(
        p.MCO1,
        p.PA8,
        embassy_stm32::rcc::Mco1Source::HSE,
        embassy_stm32::rcc::McoPrescaler::DIV1,
    );

    let mut dma_buffer = [0x00_u16; 128];
    let mut i2s = I2S::new_txonly(
        p.SPI3,
        p.PC12, // sd
        p.PA4,  // ws
        p.PC10, // ck
        p.PC7,  // mck
        p.DMA1_CH7,
        &mut dma_buffer,
        Hertz(48_000),
        Config::default(),
    );
    i2s.start();

    for i in 0_u16.. {
        i2s.write(&[i * 2; 64]).await.ok();
        i2s.write(&[i * 2 + 1; 64]).await.ok();
    }
}
```

elagil, 2025-02-02:
I don't think SPI/I2S is different on the two chips. They have the exact same register map and, at first glance, the clock sections are identical.

Probably the difference is in RCC.

vinsynth, 2025-02-02:
You're right about that--it seems the f411 has a dedicated PLLI2S divisor PLLI2SM, whereas the f401 uses the same PLLM divisor as is used for the main PLL; the `pac` must assume the latter to always be the case.

elagil, 2025-02-02:
The implementation currently lets you do forbidden things: https://github.com/embassy-rs/embassy/blob/main/embassy-stm32/src/rcc/f247.rs#L12

But even then, your configuration is not invalid.

I see that F401 and F411 use the same register layout in `stm32-data`. This one: https://github.com/embassy-rs/stm32-data/blob/main/data/registers/rcc_f4.yaml

Judging from the differences in `RCC_PLLI2SCFGR` alone, this does not seem right.

vinsynth, 2025-02-02:
Yep, looks like PLLM is never set for I2S:

https://github.com/embassy-rs/embassy/blob/ae5ad91bbb6a158971c858f69ad25ca86025f2be/embassy-stm32/src/rcc/f247.rs#L417

https://github.com/embassy-rs/embassy/blob/ae5ad91bbb6a158971c858f69ad25ca86025f2be/embassy-stm32/src/rcc/f247.rs#L393

vinsynth, 2025-02-02:
Just added a pull request for this change with some basic `#cfg(...)` included: https://github.com/embassy-rs/embassy/pull/3836#issue-2826081351; there may be some configuration cases for f7 chips, etc. that I'm not handling.

elagil, 2025-02-03:
By the way, is that a blackpill v2?

vinsynth, 2025-02-03:
It’s a v3.1 from WeAct.

Also, I’m thinking of updating the example that sparked this whole discussion so it’ll work as intended. Are all the f4 examples supposed to use the same chip?

elagil, 2025-02-03:
I'm not sure about the conventions of chips to use for examples, but it might be useful to have examples with common platforms like yours.

Do you have a DAC attached? It would be very straightforward to enable my USB audio firmware for that target: https://github.com/blus-audio/firmware-rs/tree/main/blackpill-pcm5102a

vinsynth, 2025-02-04:
In that case, I can just submit my test code as an example, and see if any issues are raised.

The board itself doesn't have a DAC on it; I'm using an external pcm5102a.

elagil, 2025-02-04:
> The board itself doesn't have a DAC on it; I'm using an external pcm5102a.

Yes, that's perfect! In case you are interested in testing it, I would like to port my sound card to your blackpill v3.1 and see if it works as expected.

> Edit: Just made a pull request: [#3845](https://github.com/embassy-rs/embassy/pull/3845)

Great! I think it's useful.

vinsynth, 2025-02-04:
Sure, I could test it--just link the source when it's ready.

elagil, 2025-02-04:
Thanks! Here you go: https://github.com/blus-audio/firmware-rs/pull/6
We can move potential discussion there too.

hazer-hazer, 2025-03-30:
Hi, I'm encountering similar issue with my STM32F412RET6. This is my i2s config:
```rust
    config.rcc.plli2s = Some(Pll {
        prediv: PllPreDiv::DIV8,
        mul: PllMul::MUL384,
        divp: None,
        divq: None,
        divr: Some(PllRDiv::DIV5),
    });
```
But outputting I2S_PLL to MCO2 gives 9.6MHz, 8 times slower than should be.