embassy-rs/embassy #3695: unresolved import `use embassy_stm32::cordic`

saosebastiao, 2024-12-29:
I can't seem to properly import the `embassy_stm32::cordic` module regardless of how I configure my project. I tried looking for feature definitions, but it seems like the cordic module should only be disabled for two specific stm32 variants:

https://github.com/embassy-rs/embassy/blob/a4f8fddd696ca2e3705827ba4b3806cbadcb3134/embassy-stm32/src/lib.rs#L59C1-L59C52

The docs also show this as available for the exact processor that I'm using: 
https://docs.embassy.dev/embassy-stm32/git/stm32g474ce/cordic/index.html

For reference, this is my dependency statement: 

```
embassy-stm32 = { version = "^0.1", features = [ "stm32g474ce","rt", "defmt", "memory-x","time","time-driver-any","exti","low-power"] }
```

yodaldevoid, 2025-01-07:
The `cordic` support was added after the 0.1.0 release. You will need to use the git repo directly to get this support.

saosebastiao, 2025-01-07:
Thank you, I just found the part in the embassy book about using git repo dependencies. I can do that. I don't know why, but that has always made me a little uncomfortable. It seems like embassy hasn't had a versioned release in quite a while. Is there a schedule or milestone-based target for version releases?

Dirbaio, 2025-01-10:
embassy-stm32 v0.2.0 is now released on crates.io

closed 2025-01-10 by Dirbaio