embassy-rs/embassy #3719: How to do GPIO-bitbanging in an embassy world?

twitchyliquid64, 2025-01-03:
Hi!

I was wondering if you had any guidance on how to do GPIO bitbanging in embassy.

#### What I was hoping for

Some kind of preemptive multitasking for really hot but tiny sections of code, such as those that read global state and set GPIOs at like 10k hz.

```rust

// Doesnt exist but i want it to
#[embassy::realtime_timer(10khz)]
async fn shift_out_data() {
  let word = bbqueue.dequeue();
  gpios.write(word);
}
```

#### What I found

It seems like embassy does support priorities using `InterruptExecutor`, but that relies on luck in terms of the higher-priority task getting airtime at the right moment. I was hoping for a way to wire a timer so it would fire exactly on some cadence, with the only jitter being higher-priority interrupts. But RP2040 crate doesn't even expose the other 3 alarms that could be used for this purpose.

Dirbaio, 2025-01-04:
> It seems like embassy does support priorities using InterruptExecutor, but that relies on luck in terms of the higher-priority task getting airtime at the right moment. 

InterruptExecutor *is* preemptive scheduling, it will preempt lower-priority tasks at any time if a higher-priority task is woken. When using InterruptExecutor you can only get latency+jitter from:

- the executor itself. Waking+polling a task adds a latency of ~100 cycles, it should add no jitter since it's constant-time
- if waking up from a timer, the latency from the embassy-time driver (varies depending on how many tasks you have in the timer queue)
- critical sections
- interrupts at higher priority.

You can also use a hardware timer and use plain old interrupts. `#[interrupt]` from `cortex-m-rt` is still usable under Embassy, just like any embedded Rust project. You can do this with the PAC directly, `embassy-rp` doesn't have any low-level timer driver at the moment, but it could have one. In this case you'd save the latency+jitter from the executor and timer queue.

twitchyliquid64, 2025-01-04:
> InterruptExecutor _is_ preemptive scheduling, it will preempt lower-priority tasks at any time if a higher-priority task is woken.

Interesting, I was under the impression that `Timer::after_ticks(1).await;` would result in the future pending after the duration has elapsed, but only getting airtime once it reached that spot on the pending queue.

Sounds like thats not the case? Somehow the timer expiring immediately wakes the task without crunching through earlier entries in the task-pending list?

Dirbaio, 2025-01-04:
There's a single global timer queue, then a per-executor run queue. (this applies to embassy-time v0.4, embassy-executor v0.7)

In theory you can get jitter from the timer queue, if lower-prio tasks spam lots of timers it can delay high-prio timers a bit. However, the timer queue only wakes tasks, which is very fast (it just places the task in the executor's run queue, it doesn't poll it yet). So it takes *a lot of timer spam* to cause problems.

Waking a high-prio task pends the InterruptExecutor's interrupt. This preempts anything that was running at low prio, and then the irq handler goes through the run queue and polls the tasks.

twitchyliquid64, 2025-01-04:
Gotcha, thanks! I had missed that there was a global timer queue, and that the higher-prio executor was given airtime immediately via software interrupt rather than the interrupt pend just being added to the run queue. So it sounds like `InterruptExecutor` is actually pretty realtime, with only a few caveats:

1. Needs to be a higher priority interrupt (which ofc it should be)
2. Needs to be okay with the ~100 or so cycles which are used for processing the timer queue and doing the software interrupt
3. Needs to be okay with jitter caused by entries higher up the timer queue or by higher priority interrupts

Which is pretty solid.

My takeaways are:

1. `InterruptExecutor` is preemptive and pretty dope for everything except the most insane timing requirements
2. Use timers + interrupts manually for everything else, if they are even possible for your clock speeds.

Thanks again! 🙏

closed 2025-01-04 by twitchyliquid64