embassy-rs/embassy #3721: Use internal pull-ups with STM32 Timer/QEI (Quadrature Encoder) Input

leehambley, 2025-01-04:
More than happy to submit a patch if it's welcome, but as I'm new to the ecosystem I wanted to report an issue and check-in first.

I'm trying to use a simple EC11 rotary quadrature encoder for input and did already find the [`timer::Qei`](https://github.com/embassy-rs/embassy/blob/main/embassy-stm32/src/timer/qei.rs) implementation, however when using that no internal pullups/downs are set. I'm breadboarding right now, so that's fine, but it'd be nice to be able to use the internal pullups since they are there already.

My main task looks something like this:

```
#[embassy_executor::main]
async fn main(spawner: Spawner) {
    let p: embassy_stm32::Peripherals = embassy_stm32::init(Default::default());

    embassy_stm32::pac::GPIOA.cr(0).modify(|r| {
        r.set_mode(0, Mode::INPUT);
        r.set_cnf_in(0, CnfIn::PULL);
        r.set_mode(1, Mode::INPUT);
        r.set_cnf_in(1, CnfIn::PULL);
    });
    embassy_stm32::pac::GPIOA.odr().modify(|odr| {
        odr.set_odr(0, Odr::HIGH);
        odr.set_odr(1, Odr::HIGH);
    });
}
```

Without spawning any tasks that works just fine, I can see the pin to to about 3V3 essentially immediately. **Note** due to really lack of docs, and lack of familiarity for myself with the ecosystem I am **less than certain** that this is the correct way to set bits on the PUPDR, especially using the ODR to pull the pin high feels incorrect when in my task I'm immediately setting that as an input pin.

The task I have is 

```
#[embassy_executor::task]
async fn encoder_input_task(
    tim: TIM2,
    chan_a: PA0,
    chan_b: PA1, // TODO: make these generic?
    _target_rpm: &'static Mutex<CriticalSectionRawMutex, RefCell<f32>>,
    _direction: &'static DirectionAtomic,
) {
    // Initialize TIM2 for encoder mode
    let qei_chan_a = QeiPin::new_ch1(chan_a);
    let qei_chan_b = QeiPin::new_ch2(chan_b);
    let encoder = Qei::new(tim, qei_chan_a, qei_chan_b);

    loop {
        Timer::after_millis(1000).await;
        let count = encoder.count();
        info!("Count: {}", count);
    }
}
```

I checked the implementation of `Qei::new_inner` and it doesn't look to interact with these registers at all, but I just think something must be going wrong.

The patch I'd like to propose would be to add a   configuration param whether channel a/b (a and b) should be pulled up or down when setting up the `Qei`.

Dirbaio, 2025-01-05:
> The patch I'd like to propose would be to add a configuration param whether channel a/b (a and b) should be pulled up or down when setting up the Qei.

Sounds good, this is how the rest of the drivers do it. A PR would be welcome.

leehambley, 2025-01-05:
PR above #3728

leehambley, 2025-01-05:
Closing this issue and moving discussion to the PR. Thanks @Dirbaio

closed 2025-01-05 by leehambley