embassy-rs/embassy #3740: stm32g0b1: USB driver does not have vbus detection

drbartling, 2025-01-07:
`usb::Driver::new()` does not take an input pin for vbus detection.
Without vbus detection we are unable to get a Disconnect error from `read_packet()`.

As application currently need to use timeouts to check for disconnection and these can't be guaranteed to give correct results at all times.

To fix this, we want to add an `Option<ExtiInput>` to the `usb::Driver::new()` function call as a breaking change.

References:
https://matrix.to/#/!YoLPkieCYHGzdjUhOK:matrix.org/$Dh-en2iZjDp-z_24twqmlorKo0lPJsk9RupKxCdHOlk?via=matrix.org&via=beeper.com&via=mozilla.org

seijikun, 2025-01-12:
Got the same problem with self-powered ESP32S3 devices.
[This](https://docs.espressif.com/projects/esp-idf/en/v5.1/esp32s3/api-reference/peripherals/usb_device.html#self-powered-device) is how that's handled in esp-idf.

drbartling, 2025-02-13:
Alternatively, could we make use of the suspen and susprdy bits?

[RM0444](https://www.st.com/resource/en/reference_manual/rm0444-stm32g0x1-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) p. 1285