embassy-rs/embassy #3747: STM32H7xx I2C Example Fails with DMA and Timeout Errors

EzraWolf, 2025-01-08:
Hello.

I spent the last month in toil trying to get RTIC and a Teensy 4.1 to simple test I2C on a handful of I2C devices. About 2 full weeks of work later, nothing. Predictably, I have similar issues here.

I am using a Daisy Seed (STM32H750IBK6) and a SparkFun BMI270 with the QWIIC connector.

SDA and SCL and pulled high via 10K resistors each, breadboard is tested and should be connected perfectly fine with 3V3 digital power.

I have absolutely no idea why this is so consistently an issue and it seems I am one of the only people with this issue besides #2884 

At the moment my STLink Debugger is missing, so I am writing over DFU, running the `build-hex` alias I have in `.cargo/config.toml` using `objcopy` to compile a hex file, which is then uploaded via the STM32CubeProgrammer.

The error received in perpetuity, no matter what, is the following:
```
ERROR panicked at 'DMA: error on DMA@40020000 channel 5'
└─ embassy_stm32::dma::dma::on_irq_inner @ /Users/a/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/embassy-stm32-0.1.0/src/fmt.rs:106 
2.000376 ERROR panicked at /Users/a/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/defmt-0.3.10/src/lib.rs:380:5:
explicit panic
└─ panic_probe::print_defmt::print @ /Users/a/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/panic-probe-0.3.2/src/lib.rs:104
```

## [Here is my code in a GIST](https://gist.github.com/EzraWolf/37900392836b68c5ad005a4a297119ce)

Any help would be greatly appreciated. I suspect this is a stupid error of my own such as incorrect DMA channels used or similar.

Dirbaio, 2025-01-08:
you're putting the `RAM` section in the linker script in DTCM. DMA can't access DTCM.

Possible solutions:
- Put `RAM` in AXISRAM. 
- Use blocking SPI, which doesn't use DMA.
- Keep ram in DTCM, but try ensuring every single buffer used for DMA is manually allocated in AXISRAM. Not recommended.

EzraWolf, 2025-01-08:
Oh my gosh you are a life saver - thank you so much. I completely overlooked this

To help others - I referenced "Table 1" on the [STM32H750IBK6 datasheet](https://www.lcsc.com/datasheet/lcsc_datasheet_2304140030_STMicroelectronics-STM32H750IBK6_C730200.pdf)

[image]

With this table of the memories, I made this `memory.x` file from the one under [`examples/stm32h7/memory.x`](https://github.com/embassy-rs/embassy/blob/cd70c19ab5652fd58ad397ccef207298d52e66aa/examples/stm32h7/memory.x). Here is the exact file I used, nothing else changed:

```
MEMORY
{
    /* Info from STM32H750IBK6 Datasheet Table 1*/
    FLASH    : ORIGIN = 0x08000000, LENGTH = 128K  /* STM32H750(VB|ZB|IB|XB) FLASH */
    RAM      : ORIGIN = 0x24000000, LENGTH = 512K  /* SRAM,  AXI Bus */
    RAM_D3   : ORIGIN = 0x38000000, LENGTH = 64K   /* SRAM4, D3 Domain */
}

SECTIONS
{
    .ram_d3 :
    {
        *(.ram_d3)
    } > RAM_D3
}
```

Lastly, where can I learn more about `memory.x` files and how to better work with them in the future?

EzraWolf, 2025-01-08:
Thank you for the link, look through the docs it seems I have a lot to learn.

closed 2025-01-08 by EzraWolf