embassy-rs/embassy #3754: AVR examples?

MarkuBu, 2025-01-11:
I noticed that AVR support is listed in the official documentation

https://docs.embassy.dev/embassy-executor/git/std/index.html#architecture

How far has AVR support progressed? Are there any examples, e.g. for an Arduino Uno?

All I found is this https://github.com/niclashoyer/embassy-avr-blink

Has there been any progress since then?

lulf, 2025-01-11:
The only thing that have been added for AVR is support in the embassy-executor. AFAIK no one is actively working on AVR support.

There is no HAL inside embassy, but you might be able to pick a HAL from another crate on crates.io (I can see some atmega PAC and HALs there). There is also a PR here https://github.com/embassy-rs/embassy/pull/3696 for adding an example.

brainstorm, 2025-03-05:
Well, I have this toy example: https://github.com/brainstorm/embassy-as5600/

And it worked well on Espressif ICs around a year ago, but for AVR I was facing some challenges (panics). Fast forward to today, it does not seem to compile (it 100% compiled at least back then!):

```
% rustc --version
rustc 1.87.0-nightly (f04bbc60f 2025-02-20)

% cargo build
(...)
error[E0432]: unresolved import `core::sync::atomic::AtomicPtr`
   --> /Users/rvalls/.cargo/git/checkouts/embassy-c08a80187403f815/0edd45e/embassy-executor/src/raw/mod.rs:31:26
    |
31  | use core::sync::atomic::{AtomicPtr, Ordering};
    |                          ^^^^^^^^^
    |                          |
    |                          no `AtomicPtr` in `sync::atomic`
    |                          help: a similar name exists in the module: `AtomicI8`
    |
    = help: consider importing this struct instead:
            portable_atomic::AtomicPtr
note: found an item that was configured out
   --> /Users/rvalls/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/src/rust/library/core/src/sync/atomic.rs:301:12
    |
301 | pub struct AtomicPtr<T> {
    |            ^^^^^^^^^
note: the item is gated behind the `ptr` feature
   --> /Users/rvalls/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/src/rust/library/core/src/sync/atomic.rs:295:7
    |
295 | #[cfg(target_has_atomic_load_store = "ptr")]
    |       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

For more information about this error, try `rustc --explain E0432`.
error: could not compile `embassy-executor` (lib) due to 1 previous error
warning: build failed, waiting for other jobs to finish...
```

/cc @autofix

brainstorm, 2025-03-05:
There are more recent (and seemingly working examples) too over here: https://github.com/simmsb/tyrfing (by @simmsb who seems to be using his own fork?).

simmsb, 2025-03-05:
> who seems to be using his own embassy fork?

Yeah, I found that 64 bit time caused a significant amount of code to be generated, switching out to a 32 bit reduced code size by quite a bit. Though perhaps there are other ways to prevent to 8 byte load/add/store instruction sequences being inlined everywhere.

On that project I spent a fair amount of time micro-optimising code size, I'll try to remember everything I did:

1. Swapping the embassy timer to a 32 bit
2. Removing the 'don't replace waker if both wake the same task' check [here](https://github.com/simmsb/embassy/commit/e9184f98538da4f2b70bf3c82fefd5b5a5d50d1c#diff-d1cc9605dbb320035bed272c0720fbfcc62bdf88509e4c61061f4d3ad0f73ac6L24-R27).
3. Using turbowakers (these save quite a lot of code size on AVR)
4. Reducing the number of different future types in use (e.g. by only using future combinators with `dyn Future`)
5. [Preventing inlining](https://github.com/simmsb/tyrfing/blob/master/src/states.rs#L417-L439) of larger async fn futures, as generated future size seems to scale nonlinear with the number of states.

leftger, 2026-11-05:
A forked project has examples. Closing for now.

closed 2026-11-05 by leftger