embassy-rs/embassy #3774: Missing entries for UART DE pins for H7RS

williams-one, 2025-01-15:
I think that the assignment for DriverEnable pins is missing for H7RS

The datasheet reports them as being shared with RTS pins
[image]

But there is no corresponding entry `pin_trait_impl!(crate::usart::DePin, UARTXX, XXX, XXX);`  in the generated data

I think that the root cause of the problem is that there is no entry for _DE pins in the .xml descriptors such as `stm32-data\sources\cubedb\mcu\STM32H7S7L8HxH.xml`
[image]

If this is an actual problem I can help with the fix, but I need some guidance since I don't know exactly where to apply the fix.  Here or in stm32-data and where to exactly put the override...

Dirbaio, 2025-01-21:
ideally we should avoid adding overrides in embassy-stm32 for bad data, so fix should be in stm32-data.

If it really is the case that all RTSs are also DEs and all CTSs are NSSs then some well-placed `if` in `stm32-data-gen` should do the trick.

williams-one, 2025-01-22:
OK.  I have also double checked the xml data from STM sources https://github.com/STMicroelectronics/STM32_open_pin_data and it looks that they have fixed the peripheral description in CubeMx 6.13 release https://github.com/STMicroelectronics/STM32_open_pin_data/commit/3e9ade95efd8e2331b8a53a500d73966f8f840ad

[image]

I have seen that stm32-data is using sources from an older CubeMx release.  I imagine that moving to a new one would mess up a lot of things in embassy since ST always keep changing things :(

But do you think that it could be possible to patch just the descriptor files for H7RS in stm32-data-sources?  Or should I go on with the manual override?

williams-one, 2025-01-22:
Oh and by the way... NSS pins are fine also in the old xml.  Only DE pins are missing

PR #4154 (2025-05-01): Update stm32-metapac
Fixes #3774
Fixes stm32h7r3r8 failing to build due to missing XSPI
closed 2025-05-01 by Dirbaio