embassy-rs/embassy #3780: Using probe-rs fails with  "got SwdDpWait/SwdApWait" when CPU sleeps a lot on STM32G4

blaa, 2025-01-16:
I believe this used to work just fine. Currently the problem is easily reproducible. With main application loop waking up every 2ms, the probe-rs can `run` code although it hits few retries:
```
bla@univdev …smartenough/io-ctrl λ cargo run --release --bin gate --features bus-addr-gate -- --preverify 
warning: unused manifest key: package.default-build
   Compiling io-ctrl v0.1.0 (/home/bla/_smarthome/smartenough/io-ctrl)
    Finished `release` profile [optimized + debuginfo] target(s) in 1.89s
     Running `probe-rs run --chip STM32G431CBUx --log-file /dev/null target/thumbv7em-none-eabi/release/gate --preverify`
 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: SwdDpWait
 WARN probe_rs::probe::stlink: got SwdDpWait/SwdApWait, retrying.
 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: SwdDpWait  <-- Some warnings, but eventually retry works.
 WARN probe_rs::probe::stlink: got SwdDpWait/SwdApWait, retrying.
 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: SwdDpWait
 WARN probe_rs::probe::stlink: got SwdDpWait/SwdApWait, retrying.
 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: SwdDpWait
 WARN probe_rs::probe::stlink: got SwdDpWait/SwdApWait, retrying.
     Finished in 0.54s
0.000000 INFO  Gate preinit
└─ gate::____embassy_main_task::{async_fn#0} @ src/bin/gate.rs:28  
0.000000 DEBUG rcc: Clocks { hclk1: MaybeHertz(144000000), hclk2: MaybeHertz(144000000), hse: MaybeHertz(8000000), hsi48: MaybeHertz(48000000), pclk1: MaybeHertz(144000000), pclk1_tim: MaybeHertz(144000000), pclk2: MaybeHertz(144000000), pclk2_tim: MaybeHertz(144000000), pll1_p: MaybeHertz(0), pll1_q: MaybeHertz(48000000), rtc: MaybeHertz(32000), sys: MaybeHertz(144000000) }
└─ embassy_stm32::rcc::set_freqs @ /home/bla/_smarthome/embassy-fdcan/embassy-stm32/src/rcc/mod.rs:71  
0.000000 INFO  DBGMCU - true true true
```

Main loop looks like this:
```
loop {                                                                                                                
    Timer::after(Duration::from_millis(2)).await;                                       
    if cnt % 3000 == 0 {                                                                
        defmt::info!("Tick: {:?}", status::COUNTERS);                                   
    }                                                                                   
    cnt += 1;                                                                                                                     
}                                                                                       
```

The DBGMCU info message above is from:
```
let config = common::config_stm32g4();         
let peripherals = embassy_stm32::init(config); 
let dbgmcu = embassy_stm32::pac::DBGMCU.cr().read();                                    
info!("DBGMCU - {} {} {}", dbgmcu.dbg_sleep(), dbgmcu.dbg_stop(), dbgmcu.dbg_standby());
```

So it seems to be set.

Changing the main loop from sleeping 2ms to 3000ms will cause first flashing to work just fine, then after ctrl-c the probe won't be able to connect:

```
bla@univdev …smartenough/io-ctrl λ cargo run --release --bin gate --features bus-addr-gate -- --preverify 
warning: unused manifest key: package.default-build
   Compiling io-ctrl v0.1.0 (/home/bla/_smarthome/smartenough/io-ctrl)
    Finished `release` profile [optimized + debuginfo] target(s) in 1.90s  
     Running `probe-rs run --chip STM32G431CBUx --log-file /dev/null target/thumbv7em-none-eabi/release/gate --preverify`
      Erasing ✔ 100% [####################]  50.00 KiB @  42.22 KiB/s (took 1s)  
  Programming ✔ 100% [####################]  49.00 KiB @  16.04 KiB/s (took 3s)   
     Finished in 3.05s         
0.000000 INFO  Gate preinit    
└─ gate::____embassy_main_task::{async_fn#0} @ src/bin/gate.rs:28
0.000000 DEBUG rcc: Clocks { hclk1: MaybeHertz(144000000), hclk2: MaybeHertz(144000000), hse: MaybeHertz(8000000), hsi48:MaybeHertz(48000000), pclk1: MaybeHertz(144000000), pclk1_tim: MaybeHertz(144000000), pclk2: MaybeHertz(144000000), pclk2_tim: MaybeHertz(144000000), pll1_p: MaybeHertz(0), pll1_q: MaybeHertz(48000000), rtc: MaybeHertz(32000), sys: MaybeHertz(144000000) }  
└─ embassy_stm32::rcc::set_freqs @ /home/bla/_smarthome/embassy-fdcan/embassy-stm32/src/rcc/mod.rs:71  
0.000000 INFO  DBGMCU - true true true                                                                                          
└─ io_ctrl::boards::ctrl_board_v1::{impl#0}::init @ src/boards/ctrl_board_v1.rs:94  
(...)
Finished with ctrl+c 
```

Second run:
```
    Running `probe-rs run --chip STM32G431CBUx --log-file /dev/null target/thumbv7em-none-eabi/release/gate --preverify --probe '0483:3748:6C65090132124647524B4E'`
 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: SwdDpWait
 WARN probe_rs::probe::stlink: got SwdDpWait/SwdApWait, retrying.
[... 60 lines cut ...]

My embassy version is:
```
commit 169f9c27aa33a279aad51a92b52fc047a54b82af (HEAD -> main-upstream, upstream/main)
Merge: 1e6b807ac ed63f8063
Author: Dario Nieuwenhuis <dirbaio@dirbaio.net>
Date:   Wed Jan 15 15:24:29 2025 +0000
```

Parts of Cargo.toml:
```
embassy-futures = { version = "0.1.1" }
embassy-sync = { version = "0.6.1", features = ["defmt"] }
embassy-time = { version = "0.4.0", features = ["defmt", "defmt-timestamp-uptime", "tick-hz-32_768"] }
embassy-stm32 = { version = "0.2.0", features = ["defmt", "time-driver-any", "unstable-pac", "time", "stm32g431cb"] }   
embassy-executor = { version = "0.7.0", features = ["arch-cortex-m", "executor-thread", "defmt", "task-arena-size-8192"] }  
(...)
embassy-stm32 = { path = "/home/bla/_smarthome/embassy-fdcan/embassy-stm32" }
embassy-time = { path = "/home/bla/_smarthome/embassy-fdcan/embassy-time" }
embassy-futures = { path = "/home/bla/_smarthome/embassy-fdcan/embassy-futures" }
embassy-sync = { path = "/home/bla/_smarthome/embassy-fdcan/embassy-sync" }
embassy-usb = { path = "/home/bla/_smarthome/embassy-fdcan/embassy-usb" }
embassy-executor = { path = "/home/bla/_smarthome/embassy-fdcan/embassy-executor" }                                                                                                          
embassy-embedded-hal = { path = "/home/bla/_smarthome/embassy-fdcan/embassy-embedded-hal" }                                                                                                                                         
```

frere-jacques, 2025-01-17:
I have the, most likely, very same issue on STM32F103.

blaa, 2025-01-17:
After a bit of fiddling I can add that this happens with probe-rs-tools 0.25 and NOT with probe-run 0.3.11. So using probe-run might be a good workaround.

Also interaction suggests it's probe-rs that puts device into some `mode` after hitting ctrl-c (or immediately after running) that makes it impossible to reach it later:

- build code with longer sleeps.
- probe-run works, ctrl-c interrupts
- probe-run works, ctrl-c interrupts
- probe-rs run works, ctrl-c interrupts
- neither probe-rs run not probe-run work.

So probe-rs kinda taints the device.

Other, but seem irrelevant:
- Rust 1.83 (forced through rust-toolchain.toml. Was required for embassy before recent release)
- rtt-target
- probe-rs-tools 0.24 doesn't build right now (fails on zerocopy crate)

blaa, 2025-01-17:
Probe-rs disables dbgmcu settings when closing in Drop implementation for Session struct:
```
impl Drop for Session {
    #[tracing::instrument(name = "session_drop", skip(self))]
    fn drop(&mut self) {
        if let Err(err) = self.clear_all_hw_breakpoints() {
            tracing::warn!(
                "Could not clear all hardware breakpoints: {:?}",
                anyhow::anyhow!(err)
            );
        }

        // Call any necessary deconfiguration/shutdown hooks.
        if let Err(err) = { 0..self.cores.len() }.try_for_each(|core| match self.core(core) {
            Ok(mut core) => core.debug_core_stop(), // <-- HERE
            Err(Error::CoreDisabled(_)) => Ok(()),
            Err(err) => Err(err),
        }) {
            tracing::warn!("Failed to deconfigure device during shutdown: {:?}", err);
        }
    }
}

/// (... stm32_armv7.rs)
    fn debug_core_stop(
        &self,
        memory: &mut dyn ArmMemoryInterface,
        _core_type: CoreType,
    ) -> Result<(), ArmError> {
        let mut cr = dbgmcu::Control::read(&mut *memory)?;
        cr.enable_standby_debug(false);
        cr.enable_sleep_debug(false);
        cr.enable_stop_debug(false);
        cr.write(&mut *memory)?;
        Ok(())
    }

```

blaa, 2025-01-25:
I'll close this. Probe-rs released new version that tries to store the register state at start, and then during disconnect recover that value instead of disabling it. It might be a problem if firmware sets the bits late and probe connects before, but that will not be frequent.

closed 2025-01-25 by blaa